UBC-Thunderbots/Software #1207: Theta* breaks tests that check obstacle intersection

jordanzliu, 2020-03-12:
### Description of the task

Theta* will return a path that has knots at vertices of obstacles, which counts as an intersection with the object. We need to differentiate between Theta* bounding boxes and the extent of the actual obstacle. 

### Acceptance criteria

- [ ] We need to differentiate between Theta* bounding boxes and the extent of the actual obstacle. 

### Blocked By

garethellis0, 2020-03-12:
Just to make sure I understand this, if the geometry objects had an `onBoundary(point)` function, then if we checked if `contains(point) && !onBoundary(point)` that would be what we're looking for?

jordanzliu, 2020-03-12:
I was thinking that Theta* would use slightly expanded bounding boxes for planning but that sounds easier to do

closed 2020-06-11 by jonathanlew