UBC-Thunderbots/Software #1210: DirectWheels Primitive is slower with new firmware

MathewMacDougall, 2020-03-13:
### Description of the task

The `DirectWheelsPrimitive` is slower as of the new firmware that introduced the app/io layers.

### Acceptance criteria

- [ ] Fix the wheel speed in `DirectWheels`

### Blocked By

jonathanlew, 2020-07-30:
Do we have an idea of why this is the case?

MathewMacDougall, 2020-07-31:
No. Possible guesses:
* We messed up something in the control when making the new firmware
* While making the new firmware we "fixed" a bug we didn't know about and the old code may have been doing "incorrect" scaling or something

I don't think this is _super_ critical or anything, but I can't say for sure if the new behavior is more correct or not.

garethellis0, 2020-08-08:
Oh, I know where this is from. Before we were (semi-)arbitrarily translating the received values to wheel speeds. When I made a change I thought that it was just wheel force (silly me) and so changed the interfaces to be a bit cleaner and actually set wheel force.

IMO we can close this.

jonathanlew, 2020-08-15:
superseded by https://github.com/UBC-Thunderbots/Software/issues/1649

closed 2020-08-15 by jonathanlew