UBC-Thunderbots/Software #1216: Convert Radio System to Protobuf

akhilveeraghanta, 2020-03-16:
### Description of the task

Our current radio system uses a custom packing system which is efficient but difficult to maintain and update as we move into the new firmware.

As part of this ticket, that protocol will need to be replaced w/ protobuf. And the necessary firmware changes on the dongle/robot and the software changes need to be field tested.

After some discussion, we decided the new packing will be as follows
`| robot id | protobuf payload | estop | `

- the robot will only de-serialize the data if the data is meant for it
- the estop is outside of the payload so that if a robot goes out of vision, it still receives the estop bit

### Acceptance criteria

- [ ] Transition old firmware to use protobuf
- [ ] Implement round robin to send packets over
- [ ] Lot of field testing
- [ ] Nuke unused code

### Blocked By
N/A

jonathanlew, 2020-05-09:
With the new sensor fusion design #1135, we've decided that we won't convert radio to protobuf immediately since we can wrap the annunciator/mrf in a radio backend and output SensorProto. I would hold off on this for now

akhilveeraghanta, 2020-05-11:
This is only for `TbotsRobotMsg`, the `TbotsMsg` would still need to be proto or custom packing for any new data we choose to send.

closed 2020-06-27 by garethellis0