UBC-Thunderbots/Software #1250: Fix Visualizer Crash on isConvex()

jonathanlew, 2020-03-30:
### Description of the task
Visualizer will sometimes crash with the ConvexPolygon constructor throwing an exception when isConvex() fails.

### Acceptance criteria

- [ ] Visualizer doesn't crash anymore

### Blocked By

MathewMacDougall, 2020-03-30:
Wait I thought this got fixed (again) or at least disabled. @jsn0716 @scveloso summoning you again (sorry)

@garethellis0 FYI

jsn0716, 2020-03-30:
My fix from #1215 did remove that error on the full system start up with `grsim` as backend, and I'm guessing a similar error is happening with the visualizer. How do you run the visualizer + reproduce the error?

MathewMacDougall, 2020-03-30:
Should be the same as running with grsim. As long as you don't provide the `--headless` flag, the Visualizer should run. Also, the Visualizer only displays stuff so if #1215 prevented these degenerate shapes from being sent, I wouldn't expect the Visualizer to fail.

@jonathanlew any chance you know where the failure came from more specifically?

jonathanlew, 2020-03-30:
yeah I have a PR coming

closed 2020-04-01 by jonathanlew