UBC-Thunderbots/Software #1267: Refine usage of loop_forever in Actions

MathewMacDougall, 2020-04-13:
### Description of the task

Created from conversation in #1261 

Some Actions like `Kick` and `Chip` currently have the option to loop forever, which doesn't necessarily make sense. We should audit our current list of actions to determine what should expose the `loop_forever` parameter or not, and what the  default should be (if any).

In the case of `Kick` and `Chip`, we will need to implement something to detect the "done" condition like if the ball has been kicked or not.

### Acceptance criteria

- [ ] For all actions, determine if they should expose the `loop_forever` parameter. If they should not, determine what the default value should be
- [ ] Implement the above changes
- [ ] Implement any new "done" conditions that are necessary

### Blocked By

MathewMacDougall, 2020-04-13:
@garethellis0 @jonathanlew @amrm3lm

MathewMacDougall, 2020-04-13:
| Tables        | Are           | Cool  |
| ------------- |:-------------:| -----:|
| col 3 is      | right-aligned | $1600 |
| col 2 is      | centered      |   $12 |
| zebra stripes | are neat      |    $1 |

amrm3lm, 2020-04-14:
very cool table

in further support for having chip/kick not loop forever, it seems that the shoot tactics already assume the kick action will be done at some point and depend on the action completing so that they themselves are done. Similarly shoot plays assume shoot tactcs will be done eventually. 

>The reason I'm unsure about the "movement related" actions is: If an Action reaches its done state (aka position), but then gets moved by an external factor, should it still try to return to its done state even if loop_forever=false? Currently, this would not be the case because of how coroutines work, hence the whole restart thing.

wouldnt the tactic that calls the action go to the next step in the tactic once the robot reaches the desired position, meaning that the move action wont be called again? 
I think this behavior could be created on the tactic level if needed

> Should it be up to the tactic to constantly restart an action if it wants? 

I think this is fair, and im guessing was the reason behind having the "loop_forever" param in the first place. especially if the cases where we want to constantly restart an action are limited (and probably only make sense to have in certain tactics).

garethellis0, 2020-04-14:
> The reason I'm unsure about the "movement related" actions is: If an Action reaches its done state (aka position), but then gets moved by an external factor, should it still try to return to its [...]

IMO no, the concept of "done" in an action should just mean that the action completed what was asked of it, not that it's maintaining some time-invariant condition. From a technical perspective, this is simpler to maintain, and fits just fine with how almost all of our logic is setup (ie. "once action is done, do this other thing immediately")

> Should it be up to the tactic to constantly restart an action if it wants? Do we even care about getting moved by external forces since this is presumably uncommon? What about the case where we want to "hold our ground" against a pushing robot?

IMO something like a "Move" action should be done once it reaches it's goal. If we wanted a "shove" behaviour, that should be a separate action with it's own done condition IMO.

> wouldnt the tactic that calls the action go to the next step in the tactic once the robot reaches the desired position, meaning that the move action wont be called again?
I think this behavior could be created on the tactic level if needed

Hmm, it depends. If the tactic is waiting to get to the right position then yep, it will move on. It seems (at least to me) a bit more elegant to just be able to wait for the action to finish, then move on. This way the determination of "done" for an action is coupled to that action, instead of us having to re-implement the "done" condition in every tactic where that action is used. 

> Also if we're looking at specific done conditions should we consider making the done() method virtual/not have it depend on the coroutine? because otherwise there is a one "call" delay between when an action is effectively done and when we actually say its done (ie when the coroutine finishes). For example the kick action yields the kickInent, but the coroutine is not done until its called again to drop out of the function.

That's a great point! Overall I think the extra effort of maintaining `done` methods for each subclass is probably not quite worth the effort / potential for bugs, especially given how simple the current check is. Definitely a good trade-off to be explicit about making though!

MathewMacDougall, 2020-04-14:
> Overall I think the extra effort of maintaining done methods for each subclass is probably not quite worth the effort / potential for bugs, especially given how simple the current check is.

This is how it worked many years ago, and indeed was difficult to maintain.

@garethellis0 seems like you're almost suggesting removing `loop_forever` from Actions entirely?

garethellis0, 2020-04-14:
Sorry, just to be clear, I think that `loop_forever` is fine, we just need to be selective about how we use it (see your table above). What made you think was suggesting we remove it entirely?

MathewMacDougall, 2020-04-14:
> Hmm, it depends. If the tactic is waiting to get to the right position then yep, it will move on. It seems (at least to me) a bit more elegant to just be able to wait for the action to finish, then [...]

Based on my table, the only Actions we would consider keeping `loop_forever` for are the "movement related ones", which you made sound like wasn't needed since generally the tactics wait for actions to finish then move on immediately.

We can start my making the changes in the table, unless anyone has suggestions for other changes. This is less work than removing `loop_forever` entirely to start. We can see how our use cases develop.

garethellis0, 2020-04-14:
Fine by me, the suggested changes in the table seem entirely reasonable to me, at least for now.

amrm3lm, 2020-04-15:
sounds good, i'll start making the changes mentioned in the table

jonathanlew, 2020-04-15:
> IMO no, the concept of "done" in an action should just mean that the action completed what was asked of it, not that it's maintaining some time-invariant condition. From a technical perspective, [...]

Do you mean that `Tactic`s would/could be responsible for maintaining some time-invariant condition.

garethellis0, 2020-04-15:
All I meant to say there was that Action's should _not_ be responsible for maintaining some time-invariant condition, not saying anything about what Tactic's should / should not do.

amrm3lm, 2020-10-07:
do we still want `intercept_action` to not have `loop_forever` set as `false` by deafult? I ask because recently in #1619 we modified `shoot_goal_tactic` to use `intercept_action` with `loop_forever=true`
@jonathanlew

jonathanlew, 2020-10-27:
@amrm3lm loop_forever is true but since done() is never called in that situation, it doesn't matter

PR #1807 (2020-11-07): Refine usage of loop forever in kick/chip/spinningMove actions

## Please fill out the following before requesting review on this PR

### Description

Changes to `kick_action` , `chip_action`, and `spinningmove_action`  made from [this](https://github.com/UBC-Thunderbots/Software/issues/1267#issuecomment-613071669) comment regarding the use of `loop_forever` in actions. 

### Testing Done

everything builds and test pass
### Resolved Issues

resolves #1267 
### Length Justification

### Review Checklist
closed 2020-11-07 by amrm3lm