UBC-Thunderbots/Software #1276: Add fields in FriendlyRobot and Converters from TbotsRobotMsg proto

jonathanlew, 2020-04-16:
### Description of the task
From the framework created in #1164, we need to add fields to represent the useful data contained in TbotsRobotMsg as designed in #1272. We also need to be able to take protos and create or update FriendlyRobots. 

### Acceptance criteria

- [ ] Add fields to FriendlyRobot
- [ ] Create update or create functions for FriendlyRobot from TbotsRobotMsg proto
- [ ] Unit testing

### Blocked By
#1164 #1272

MathewMacDougall, 2020-04-16:
Just to clarify, what info are you thinking of including in the `FriendlyRobot` or `Team` classes? IMO we should avoid putting anything unnecessary for AI logic into the `Friendly...` classes. In the case of #1277 , it should subscribe to the `RobotStatus` messages directly to get the additional data it needs.

jonathanlew, 2020-04-16:
> Just to clarify, what info are you thinking of including in the `FriendlyRobot` or `Team` classes? IMO we should avoid putting anything unnecessary for AI logic into the `Friendly...` classes. In [...]

With the new SensorFusion architecture #1135, there won't be any `Subject`s that output RobotStatus. Visualizer can `Observe` either `SensorProto` or `World`

jonathanlew, 2020-04-16:
> With the new SensorFusion architecture #1135, there won't be any `Subject`s that output RobotStatus. Visualizer can `Observe` either `SensorProto` or `World`

From offline discussion, Visualizer will `Observe` both `SensorProto` or `World`, at least for now

jonathanlew, 2020-07-05:
We are opting for a different architecture, see #1477 and #1522 for more details

closed 2020-07-05 by jonathanlew