UBC-Thunderbots/Software #1290: Ball Possession Improvement

akhilveeraghanta, 2020-04-19:
### Description of the task

Currently in `/software/ai/evaluation/possession.cpp`, we have functions that evaluate whether or not a team has possession of the ball. 

This is achieved through looking at vision data which has information about the robot position and the ball position, and checking if the ball is near a robots dribbler (in which case it has "possession").

We currently also check which robot can intercept the ball the fastest, and return that as the robot with the "effective possession". If there is no interception, we fall back to the closest robot to the ball. This logic can use improvement to have a more accurate sense of possession.

 

### Acceptance criteria

- [ ]  Read [MCT Susano Logics 2020 TDP](https://ssl.robocup.org/wp-content/uploads/2020/03/2020_TDP_MCT_Susano_Logics.pdf), section 2.2 to get an idea of what algorithm they implemented to determine possession. 
- [ ] See how we can apply those concepts to our current system
- [ ] Implement a new algorithm/update the existing one to improve possession
- [ ] Develop tests

### Blocked By
The break beam is a sensor underneath the dribbler of the robot that gets tripped when a ball is caught in the dribbler, which can be used as additional information to determine possession. 

Previously this information was not propagated up to AI, but that will change once the sensor fusion layer is implemented.

So technically, this ticket is kind of blocked by sensor fusion , we only care about the break beam data being a part of the world. But that easily be another ticket once sensor fusion is implemented. But just keep in mind that the world will eventually have information about friendly robot break-beams.

akhilveeraghanta, 2020-04-19:
There was a bit of discussion on where this should live:
Here is the slack thread:

Mathew MacDougall  12:10 PM
I think the main question is if it should live in sensor fusion of the AI. The algorithm MCT uses is stateful, meaning we'd likely want to treat it as a filter/class that gest updated that we can query for the latest possession data
12:11
It's definitly more of an "AI concept" so would make sense to live there, partially because it also makes testing slightly easier. If it was part of SensorFusion we'd need to include that posession info in the World somehow, which I'm not sure if we want. Thoughts?

Jonathan Lew  12:17 PM
I think that keeping it as an evaluation function in AI makes a lot of sense and AI can be stateful, for example how we deal with previous robot states

Akhil Veeraghanta:mathew_satisfied:  12:20 PM
Id argue it makes more sense to live in sensor-fusion because we would use the alg to determine possession for enemy robots based on vision data and use the break-beam to detect possession for friendlies.
once it reaches AI is doesn’t matter how it was determined (vision/breakbeam), we just know which robot has possession (edited)

MathewMacDougall, 2020-04-19:
A couple additional notes from me
* Yes technically breakbeam data is useful but this algorithm should be able to work without it, as we don't always have that data.
* We also have functions like `teamPassInProgress` which IMO is a bit "hacky" or at least fragile. Ideally we can find a more "stable" or generalization solution that won't require checking so many edge cases.
* Our current `teamHasPossession` function is currently stateless. The algorithm proposed my MCT is stateful (IIRC). I think we can choose whatever works better, and can adapt the system to use the function as needed (ie. stateless would be the same, stateful could be an injected class or something). Just pointing it out.

jonathanlew, 2020-04-21:
A couple notes from me about how I imagine the interface to work:
* a "filter" in the sensor fusion 
* it will take a SensorProto (#1278) and update some field(s) (which you can define) in World. It will keep internal filter state in a filter object within SensorFusion. For example, a function might look like `PossessionState BallPossessionFilter::filter(SensorProto proto)`
* evaluation functions that use PossessionState to determine who has the ball

jonathanlew, 2020-09-18:
Closing for now, since there is other work that make it redundant https://github.com/UBC-Thunderbots/Software/pull/1722

closed 2020-09-18 by jonathanlew