UBC-Thunderbots/Software #1308: Create a simple GUI for the XBox control

MathewMacDougall, 2020-04-22:
### Description of the task

Create a simple GUI for when we control the robots with an xbox / handheld controller. This should only need a DynamicParameter widget to control the controller (see #1307 ), and a status to show robot status / information.

### Acceptance criteria

- [ ] Create a simple GUI for the handheld controller

### Blocked By

jonathanlew, 2020-09-28:
The `DynamicParameterWidget` currently shows all dynamic parameters as it by default uses `MutableDynamicParameters`. For this application, that's too much because we only want the ability change some relevant parameters, i.e. handheld controller parameters. We can make small tweaks to the interface to allow it to accept child configs so that we can pass in a subset of the config tree.

akhilveeraghanta, 2022-02-05:
moving away  from xbox  controller for a web-based controller running on the jetson nano

closed 2022-02-05 by akhilveeraghanta