UBC-Thunderbots/Software #1313: Restructure input vs output folders in backend

jonathanlew, 2020-04-25:
### Description of the task
Currently, the backend folder is split between input and output, with the idea that input is data coming into the software stack and output is the data that's sent out from the software stack. This is not actually the case, however, with radio_output also serving the role of receiving data from the robots. 

Some ideas include:
* separating out input from radio_output
* restructuring input output folders in backend to some other division, such as ssl msgs vs robot communication

Some thoughts:
* the ProtoMulticastListener can listen to TbotsRobotMsgs so that can all fit in the input folder
* mrf can't be neatly separated on an input/output basis since the annunciator (receives robot msgs, so it's an input) is a part of radio_outpu

### Acceptance criteria

- [ ] Fix the inconsistency
- [ ] Verify that nothing functional changes

### Blocked By

MathewMacDougall, 2020-04-25:
Random thoughts:
* "New" networking classes like the `ProtoMulticast...` stuff is probably self-contained enough (and will be used in other places like the Simulator) to warrant being in its own `networking` folder or something rather than in the backend.
* Because that basic networking is a lot of what the backend(s) are currently, maybe it does make sense to break it up a different way? Except for the radio/annunciator stuff, a backend would basically be an aggregation of a few `ProtoMulticast...` objects, so maybe best to split another way.
* (if the radio stuff didn't exist, the backends would _literally_ be a composition of networking objects, which I think is totally fine which is why I think a different split might be good, because it would be easy to just nuke the radio folder once it's not supported and it wouldn't require another re-organization)

jonathanlew, 2020-04-25:
> Random thoughts:
>
>     * "New" networking classes like the `ProtoMulticast...` stuff is probably self-contained enough (and will be used in other places like the Simulator) to warrant being in its [...]

We could have network, radio, grsim, and simulation folders within backend

MathewMacDougall, 2020-04-25:
> We could have network, radio, grsim, and simulation folders within backend

But there would be duplication if we did it that way? Simulation (if standalone) just uses networking, and otherwise doesn't have its own "communication code".

Also, if the simulator moves to its own top-level folder (which I'm planning on) and uses networking, then technically networking should be a top-level folder (according to our rough rules about where code is relative to where it's used).

jonathanlew, 2020-04-25:
> But there would be duplication if we did it that way? Simulation (if standalone) just uses networking, and otherwise doesn't have its own "communication code".
>
> Also, if the simulator moves to its [...]

Yes that's right. Then all that would be left is robot communicators for radio and wifi

MathewMacDougall, 2020-04-25:
that seems reasonable. Looks like there will be some duplication between the `ProtoMulticast...` classes and the current `NetworkMedium` but we can sort that out later (it's not that relevant for folder layout)

PR #1576 (2020-07-08): Integrated SensorFusion into Full System

## Please fill out the following before requesting review on this PR

### Description
* Integrated SensorFusion into AI and Backend Observers
* Deleted NetworkClient
* Rearranged Backend folder
* Removed RobotStatus from all GUIs

### Testing Done
confirmed it works visually

### Resolved Issues
resolves #1135
resolves #1313
resolves #609

### Length Justification
YEET

### Review Checklist

**_It is the
closed 2020-07-08 by jonathanlew