UBC-Thunderbots/Software #1338: Create function to check when kicks have taken place

MathewMacDougall, 2020-05-05:
### Description of the task

We need to implement a "done condition" for kicking and chipping actions/tactics, as discussed in #1321 

We have implemented an approximation of this in the `PasserTactic`. Once it is trying to pass, it continues to do so until
1. The ball's orientation of the ball's velocity vector is nearly the same as the passing robot
2. The magnitude of the ball's velocity is above some threshold

A software approximation is the best we can do for now as we don't have access to extra firmware information yet (like how long since we the kicker was fired).

We should be able to break this out into its own function that can replace the implementation in the `PasserTactic`, that we can then use for #1321 

### Acceptance criteria

- [ ] Implement a function to detect when kicks / chips have taken place
- [ ] Unit tests
- [ ] Use this function in place of existing logic in the `PasserTactic`

### Blocked By

MathewMacDougall, 2020-05-05:
@jonathanlew @garethellis0 if you have any thoughts on how this should be done.

garethellis0, 2020-05-05:
Yeah, I think a simple comparison of the ball direction vs. the expected kick direction combined with some magnitude tolerance should do the trick.

jonathanlew, 2020-05-05:
> 
> 
> Yeah, I think a simple comparison of the ball direction vs. the expected kick direction combined with some magnitude tolerance should do the trick.

In the future we could even incorporate robot sensors via sensor fusion, but this should work for the time being

PR #1732 (2020-10-05): Software approximation to check successful kicks

## Please fill out the following before requesting review on this PR

### Description
Created a function to check when kicks have taken place. 
Replaced the implementation in `passer_tactic` to use this function. 

### Testing Done
Current tests pass. 
Added unit tests for the function. 

### Resolved Issues
resolves #1338 

### Length Justification

### Review Checklist

**_It is the reviewers res
closed 2020-10-05 by amrm3lm