UBC-Thunderbots/Software #1343: Simulator: Figure out how to backspin chip

MathewMacDougall, 2020-05-14:
### Description of the task

Backspin chipping is something we may want to make use of that we don't get "for free" in simulation because we simulate in 2D. We need to figure out how to simulate backspin chipping.

### Acceptance criteria

- [ ] Figure out how to simulate backspin chipping

### Blocked By

EvanMorcom, 2020-05-19:
My first though is using the dribbler RPM to get a contact velocity between the ball and the dribbler. There is a constant ratio between the angular velocity of the dribbler and ball that is (R_dribbler/R_ball)*W_dribbler. Use that and a formula for force of a spinning ball in the air.

jonathanlew, 2022-01-18:
we will be moving to er force simulator so we'll make a new ticket if this is a priority

closed 2022-01-18 by jonathanlew