UBC-Thunderbots/Software #1369: Verify that World::updateRefboxStage and World::updateRefboxGameState works as intended

jonathanlew, 2020-05-23:
### Description of the task
Verify that World::updateRefboxStage and World::updateRefboxGameState works as intended with SimulatedTests

### Acceptance criteria

- [ ] Add SimulatedTests
- [ ] Make sure things work

### Blocked By
#1135

MathewMacDougall, 2020-05-23:
We might need to have a quick talk about what datastructure we want to use to wrap `refbox` and `gamecontroller` stuff (if we don't have that already). We'll want an "easy" way to set things things for tests.

jonathanlew, 2020-06-03:
`RefboxStage` and `RefboxGameState` are both enums. What did you have in mind?

MathewMacDougall, 2020-06-04:
Ah never mind that should be fine. Was thinking of setting the "end result" like `isHalt()` directly, but that's not really necessary.

PR #1635 (2020-08-01): Refactor GameState and add additional world tests

## Please fill out the following before requesting review on this PR

### Description
- Make fields in GameState private
- Added tests to worldtests for updateRefboxGameState and updateRefboxStage

### Testing Done
CI

### Resolved Issues
Resolves #1369 
Resolves #1481

### Length Justification

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been
closed 2020-08-01 by LiCody