UBC-Thunderbots/Software #1382: Create A Spline/Polynomial Path Planner

garethellis0, 2020-05-27:
### Description of the task
The new on-robot controller requires that we pass a polynomial to the robot as a path to be tracked. To effectively use this we must have a path planner that generates such a polynomial.

### Acceptance criteria

- [ ] Create a Polynomial2d class and move geometry stuff around to fit in in the current geom architecture
- [ ] Either fully implement the placeholder `Spline` class or transition to having the path planner just return a 2d polynomial
- [ ] Update the existing path planners and managers to fit with the effects of decision between `Spline` and just a `Polynomial`

### Blocked By

garethellis0, 2020-10-17:
## Current State
OK, so at present we have a `CubicBezierSpline2d` class that can be setup to pass through a series of points with an initial and final direction vector and generate polynomials. Now we need to create a path planner that can use this to generate 2D 3rd order polynomials that firmware can track.

## High Level Idea
At a high level, this is a constrained optimization problem, where the constraints are:
(a) Start point (current robot location)
(b) End point (final robot location)
(c) Start direction (but not magnitude) (current direction/speed of the robot) (_note, we don't know the_)
(d) The trajectory planner must be able to track whatever trajectory we come up with
(e) The trajectory must not intersect obstacles 

 If we say our `CubicBezierSpline2D` is defined by: `[start_x, start_y, start_dir_x, start_dir_y, intermediate_pt_x, intermediate_pt_y, final_x, final_y, final_dir_x]`. Since we already know what some of these values have to be, we can then we can phrase this as an optimization problem where we're trying to find some vector `x = [start_vector_mag, intermediate_pt_x, intermediate_pt_y, final_dir_x, final_dir_y]` that can be executed by the `TrajectoryPlanner` and does not intersect any obstacles.

With just the above information, we could just keep randomly generating `x` vectors until we get something that the trajectory planner can track, but this is a rather naive approach that not be performant, to say the least. To do something more intelligent we should have a cost function that can tell us how "good" a given trajectory is (ie. how close does it come to satisfying our constraints). Once we have this cost function, we can tell if changing a given value makes `x` better or worse, and thus if we should go in the other direction. So our cost function will probably look something like `f(x) = d(x) * e(x)`, where `d(x)` and `e(x)` are both functions that get smaller as `x` gets better at satisfying constraints (d) and (e). In addition, we should use heuristics to figure out what a "good"/"sane" initial vector for `x` should be.  

For constraint (e) it's not too hard to come up with a cost function, something naive like `0.1 / (min(0.1, closest_distance_to_obstacle) + 1e-7)` would probably work. For (d) things are a little harder, as the `TrajectoryPlanner` currently only returns an error if the trajectory is not possible, but the result is binary, we don't know by how much we violated the constraints. So some updates to the `TrajectoryPlanner` will be needed to return a cost or set of costs that indicate by how much a given trajectory violates the physical constraints that the `TrajectoryPlanner` enforces internally. 

Once we've got this cost function and a heuristic for generating an initial `x`, we can use our existing implementation of gradient descent to compute a final `x` that satisfies all the constraints, or hopefully at least comes close. You can tradeoff which constraints you "care" about more by scaling different factors in the cost fuction.

### A Note On Smoothness
Several of the functions in the cost function proposed here are not smooth (such as `min`), and the costs returned by the `TrajectoryPlanner` are almost certainly not smooth. On the initial implementation I wouldn't worry about this, but it's something to be aware of as a potential source of issues later down the road. 

@MathewMacDougall @jonathanlew @akhilveeraghanta

MathewMacDougall, 2020-10-17:
* High level constraints *

> Start direction (but not magnitude) (current direction/speed of the robot) (note, we don't know the)

We don't know what? Don't leave me hanging :scream_cat: 

* third paragraph*
> With just the above information, we could just keep randomly generating x vectors until we get something that the trajectory planner can track, but this is a rather naive approach that not be performant, to say the least

It may be worth investigating a "very dumb" generation scheme first just to see. When James experimented with this a few years ago, he fixed basically everything except for the intermediate point, and generated 40-50 spines with varying intermediate points, then took the best valid one. IIRC it was very fast. I think that it would be reasonable to generate something in the order of 100 splines and have it perform reasonably well (don't forget this has to scale across 6 robots, in the future up to 11). We can be pretty coarse in the search space because there's no major need to micro-optimize how close we get to obstacles just to shave a fraction of a second of the path time (for example). At the very least this may be useful for warm-starting an optimization approach, but that would be something for later.

Also please explain what (d) and (e) are cost functions for. It's implied in the implementation later, but you're not mentioning them by name.

**Overall**
`CubicBezierSpline2d` can represent a spline with arbitrarily many points. @garethellis0 's notes above only talk about this planner as generating a spline with 3 points (start, intermediate, end). The reason we are not immediately generalizing this to optimize larger splines is because we don't know if it's worth it: We don't know if it will help us find paths we otherwise could not, and it will increase computational cost.

garethellis0, 2020-10-17:
> We don't know what? Don't leave me hanging 🙀

Whoops, edited/removed that bit.

jonathanlew, 2020-10-21:
I think the hardest challenge will be making this performant. We need a super good heuristic, so that we don't have do many iterations of the optimization to reach an acceptable path. The cost function is likely to be very expensive to run, because the vector space (number of constraints) is large and it's difficult to precompute/cache all the costs we need

jonathanlew, 2022-01-18:
closing due to new motion planning project direction

closed 2022-01-18 by jonathanlew