UBC-Thunderbots/Software #1401: Remove isValid() from Field and disallow the creation of degenerate fields

MathewMacDougall, 2020-05-30:
### Description of the task

#1251 added the `isValid()` function to `Field` to check for zero-length fields because we did not want to publish `Worlds` without a valid field.

#1385 then removed the default constructor from `Field` because there were several instances where we were just manually "default constructing" fields with all zeroes.

What this has led to is the easy ability to create degenerate fields that cause later problems for consumers. It's an anti-pattern to allow invalid Filds to be created, and provide an `isValid()` function that we _trust_ people to check before doing anything with a field.

To remedy this, we should
* Delete the default constructor for `Field`
* Remove the `isValid()` function
* Throw an exception if a field is created with any zero-length parameters

Any code that was default-constructing fields unnecessarily is likely a smell that something is being given more data than it needs (eg. being given the World when all it needs is the ball / robots).

### Acceptance criteria

- [ ] Delete the default constructor for `Field`
- [ ] Remove the `isValid()` function
- [ ] Throw an exception if a field is created with any zero-length parameters

### Blocked By

MathewMacDougall, 2020-05-30:
@jonathanlew @akhilveeraghanta @garethellis0 FYI

jonathanlew, 2020-06-03:
Would be nice if it's possible...

PR #1413 (2020-06-06): Removed default World and Field Constructors

## Please fill out the following before requesting review on this PR

### Description
* Removed `World` constructor
* Removed `Field` constructor
* Make everything still work
* Removed `Field::isValid`
* Removed `RobotExpiryBufferMilliseconds` dynamic param

### Testing Done
CI

### Resolved Issues
resolves #1401 
part of #1361 

### Length Justification
Not a lot of new code

### Review Checklist

**_
closed 2020-06-06 by jonathanlew