UBC-Thunderbots/Software #1402: Update Code Conventions to Prevent Negative Loop Exit Conditions for unsigned iterator

jonathanlew, 2020-05-31:
### Description of the task

### Acceptance criteria

- [ ] Item A
- [ ] Item B
- [ ] Item C

### Blocked By

jonathanlew, 2020-05-31:
@MathewMacDougall @garethellis0 @akhilveeraghanta Any feedback on proposed solution?

garethellis0, 2020-05-31:
I'm really not sure how I feel about this. On one hand this is definitely a real problem, but on the other I really don't want to start requiring `static_cast` in `for` loops, as I feel that is both difficult to enforce and cumbersome for cases where it's not needed. 

More generally, I don't think it's reasonable to put these sort of super specific things in the coding conventions just because I think it's very unlikely that anyone will remember them or check them.

This very much feels like the exact sort of edge case that we should be covering in unit tests, but again, I totally get that it's unlikely people will always remember to. 

Sorry for the lack of concrete suggestions here, just some points I think are worth thinking about.

jonathanlew, 2020-05-31:
> I'm really not sure how I feel about this. On one hand this is definitely a real problem, but on the other I really don't want to start requiring `static_cast` in `for` loops, as I feel that is both [...]

This is the sort of thing that will slip through unit tests, because people assume their loops work as intended. 

This sort of loop should be rare because we can usually use `for each` loops. 

Might be worth scrapping the warning altogether if we can't enforce a standard that prevents infinite loops.

MathewMacDougall, 2020-06-01:
I agree something this subtle will slip through unit tests, but I think would also slip through most code review. Plus, most people aren't familiar enough with the style guide to follow this consistently (or at all) IMO.

I'm also hesitant to impose such a cumbersome convention on something so specific for similar reasons to gareth. I think it's something that will be really hard to stick to, and will be pretty much entirely on you two to enforce manually.

This is something that could maybe be caught by custom linting rules. If we gather enough valid use-cases, perhaps it's worth using / implementing a _simple_ C++ linter and integrating it into our pipeline. That way it's automated and more "well-defined". Obviously there's technical complexity and overhead with custom linting too, so just an idea.

jonathanlew, 2020-06-01:
The convention can be simple: always write `for` loops with `int`

garethellis0, 2020-06-01:
I haven't poked at it that much, but I imagine there is a reason that the compiler produces a warning for this behaviour (and why that would require `static_cast` to fix). I really don't think it's feasible to enforce that sort of convention. 

Overall this might be worth a call since it seems we're a tad deadlocked on this.

jonathanlew, 2020-06-05:
After offline discussion, we decided against adding a convention because it would be too difficult to enforce

closed 2020-06-05 by jonathanlew