UBC-Thunderbots/Software #1409: Develop integration tests for WifiBackend

akhilveeraghanta, 2020-06-01:
### Description of the task

The Wifi Backend has many moving parts, it would be nice to be able to reliably test that all callbacks have been triggered correctly, and the timeouts are working as expected. This ticket involves writing integration tests to validate functionality.

We should be able to achieve this by joining a multicast group on `loopback`, which will allow us to send/receive packets on the same computer. We can then create virtual "NetworkRobots" which simply send `RobotStatus` at a specified rate.

### Acceptance criteria

- [ ] Create TestUtil to spawn mock TestNetworkRobots to test WifiBackend
- [ ] TestNetworkRobots must be able to send RobotStatus at a specified rate, and timeout on request
- [ ] The integration tests must validate proper generation of the SensorMsg on robot timeout
- [ ] They must also validate proper functionality (i.e sending primitives on the right channel, etc...) 

### Blocked By
#1408 
#1407

MathewMacDougall, 2020-06-01:
Would this work in CI?

akhilveeraghanta, 2020-06-02:
yeah it should, every linux computer has a loopback interface, if not, they can be disabled and run locally

closed 2021-01-26 by akhilveeraghanta