UBC-Thunderbots/Software #1410: Create test framework for testing firmware primitives

MathewMacDougall, 2020-06-01:
### Description of the task

#1395 fixed an issue that we would like to have a regression test for. However in order to unit test firmware primitives, we need to mock an entire firmware_world and all its components.

Ideally, we can do the bulk of this work in a base class with a base test fixture. Then specific tests can override any values they need to for the test without having to redefine everything..

Since we need to provide C-style function pointers we could either use static functions like some other tests, or need to mock the functions another way. Since it is difficult to override / hide static functions in extended test fixture classes, we have chosen to use [fff (the fake function framework)](https://github.com/meekrosoft/fff) to achieve this.

### Acceptance criteria

- [ ] Make a base test fixture class that creates a firmware world with sane default values
- [ ] Allow other tests to extend the test fixture and override only the values / functions they need to for their respective test.

### Blocked By

MathewMacDougall, 2020-08-07:
Note: There may not be enough benefit in unit testing firmware primitives for this to be worth it. It's possible that simulated tests that just test primitives may be more useful and cover nearly all the same cases.

@jonathanlew @akhilveeraghanta thoughts?

jonathanlew, 2020-08-07:
I think it's worth keeping this ticket floating around until we write these simulated tests, if we end up doing that

PR #1790 (2020-11-14): Create a Testing Framework for Primitives
## Please fill out the following before requesting review on this PR

### Description

Mocking functions so we can easily test them in firmware primitives. Done for `autochip_move` primitive and upon inspection, there's little reason it shouldn't work for most of the other primitives.
Testing fixture for the world found in `test_util_world.h`.

### Testing Done
`autochip_move_primitive_test` uses the `fff` library to mock functions for `robot` and `world` components. `app_dribbler_setSpeed` and `app_chicker_enableAutochip` are mocked and checked if arg [... cut]
closed 2020-11-14 by jonathanlew