UBC-Thunderbots/Software #1411: Stress test Simulator with Slow AI

jonathanlew, 2020-06-03:
### Description of the task
From some unrelated work, the Simulator may behave badly if the AI ticks are very slow. For example, simulated tests may time out. If visualizer is enabled, the visualizer may crash. The task here is to find the limit for how slow AI can be before the simulator breaks and add a test for that.

### Acceptance criteria

- [ ] Find breaking point for AI
- [ ] Check if there are more graceful ways for the simulator to fail
- [ ] Add a test to provide good feedback if the AI is getting too slow
### Blocked By

MathewMacDougall, 2020-06-04:
Whoops didn't mean to close. This issue _should_ be fixed in the simulator upgrades since we're planning to run stuff in a loop rather than as threaded observers. Then we won't depend on wall-clock timeouts, which are already annoying if you want to pause at a breakpoint. Will update later once changes are done, just pointing it out.

MathewMacDougall, 2020-06-30:
This has been fixed with the new simulated test structure. Visualizer will also not crash with a slow AI.

closed 2020-06-30 by MathewMacDougall