UBC-Thunderbots/Software #1444: Change all geom functions to return const references

jonathanlew, 2020-06-07:
### Description of the task
This will reduce unnecessary copying if the caller assigns it to a const type or uses it in a const context (e.g. `for (const auto& g: getGeoms())`).  

For example, change `Point CrossShape::getMidPoint() const` to `const Point& CrossShape::getMidPoint() const` :crossed_swords: 

### Acceptance criteria

- [ ] As titled

### Blocked By

MathewMacDougall, 2020-06-08:
Good idea. Possibly applicable to a lot of our code actually.

jonathanlew, 2020-06-08:
Might not be worth it since it can sometimes degrade performance. We need to be careful

MathewMacDougall, 2020-06-08:
Oh? In what cases can performance degrade?

jonathanlew, 2020-06-08:
I haven't really looked into it, but I think it's if we're assigning a const to non-const or the other way around, and that triggers a copy. I could be that it's just for `Point`, which has copying defined in the `operator=`

jonathanlew, 2020-06-09:
Returning local variables by const reference is the same as returning a pointer to the stack, which is a big nono

MathewMacDougall, 2020-06-09:
Ah true

PR #1618 (2020-07-18): Geometry Clean Up and Folder Restructure

## Please fill out the following before requesting review on this PR

### Description
Cleaned up the last bit of geom/util, miscellaneous geom, and moved folders around

### Testing Done

### Resolved Issues
resolves #959
resolves #813 
resolves #1593 
resolves #1494 
resolves #1444

### Length Justification
Hard to break this up

### Review Checklist

**_It is the reviewers responsibility to also make sur
closed 2020-07-18 by jonathanlew