UBC-Thunderbots/Software #1475: 2019 Fleet Retirement Clean Up

jonathanlew, 2020-06-16:
### Description of the task 
When we retire the 2019 fleet, we can remove a lot of code that supported the old robots, including radio, firmware, dongle. 

### Acceptance criteria

- [ ] Delete firmware/boards/legacy_robot_stm32f4
- [ ] Remove stm32f4 targets from travis
- [ ] Add warnings_as_errors feature to stm32 in cc_toolchain/cc_toolchain_config.bzl
- [ ] Remove radio backend and mrf code

### Blocked By

akhilveeraghanta, 2020-06-17:
hmm did we want to do this? thought the plan was to support old robots so we have a reference point, and then drop support

jonathanlew, 2020-06-17:
I needed to link an issue for #1473

MathewMacDougall, 2020-06-18:
> hopefully not too distant future

plz :pray:

jonathanlew, 2022-02-08:
resolved by #2464

closed 2022-02-08 by jonathanlew