UBC-Thunderbots/Software #1491: Simulator: Show how fast the simulation is running

MathewMacDougall, 2020-06-19:
### Description of the task

Fro #1488 It would be good to have a way of visually indicating how fast the simulator is running / if it's keeping up with real-time. This could be like an FPS counter for various games / applications, for example.

This ticket is to add a way of exposing this timing data and displaying it in the simulator GUI

### Acceptance criteria

- [ ] Display simulation speed

### Blocked By

jonathanlew, 2020-07-30:
Any recommendations on how this can be done? What do you think about using the timestamp on PrimitiveSetMsg?

MathewMacDougall, 2020-07-31:
Few options and suggestions

First question is, what exactly do we want the output to look like?
1. an "FPS" counter for how often the Visualizer is rendering new data (would be a multiple of the internal timer, so would likely jump between 60 and 120 FPS for example)
2. How much new data is being received a second, in Hz. Eg. the Visualizer is receiving 74 messages a second. This would be analogous to the FPS of the overhead camera.
3. The period with which messages are sent?
4. A percentage value compared to some expected speed? Eg. 50% means we're running half as fast as real-time
5. Something else?

Second question:
* To clarify, this is for the speed of the simulation / simulator, **_not_** the AI correct?

Third question:
* I'm pretty sure this should use wall-clock time rather than timestamps, because if we are not running in real-time simply looking at the message timestamps wouldn't let us see that because the simulator will publish them with the same period due to its determinism.

Given that IIRC there was another ticket / discussion for showing how fast data is moving around the _entire_ system, we may be able to provide a foundation for all that introspection as well **_if_** we go with option 2 from the first question. It should be pretty easy to add this functionality to the Observer class so we can just query the observers for how fast they are receiving data.

How this may look (in an Observer) is storing the wall-clock time in a buffer each time a message is received. This buffer can be small (eg. 5-10 elements) so we can average the differences in time and output a slightly more stable Hz value.

PR #1746 (2020-10-03): Added Data Received Per Second Counter to Full System Main

## Please fill out the following before requesting review on this PR

### Description
- added timing functionality and unit tests to the observer class
- added a new buffer for timing data into `threaded_full_system` 
- added a data received per second counter to the visualizer

### Testing Done
tested with sim

### Resolved Issues
resolves #1491 

### Length Justification

### Review Checklist
closed 2020-10-03 by LiCody