UBC-Thunderbots/Software #1492: Visualizer: Display robot charge state

MathewMacDougall, 2020-06-19:
### Description of the task

The visualizer should display whether robots are charged, floating, or discharged.

### Acceptance criteria

- [ ] Show robot charge-state in the Visualizer

### Blocked By

jonathanlew, 2020-06-20:
This should be generic so RobotDiagnostics can use it too

MathewMacDougall, 2020-06-20:
RobotDiagnostics doesn't show the robots in a "camera" view though, so I think these would be different.

MathewMacDougall, 2020-08-07:
More meta discussion, do we actually think this is a useful feature? It would likely manifest as displaying the capacitor voltage (since we can't necessarily read back the charge "mode" from the robot).

Now that I think more about it I made this ticket because our old AI showed this info, but didn't think of why we need it. For nearly all gameplay-related usages / robocup I don't actually see this info being super helpful, because there's no real reason for us to see cap voltage beyond error reporting.

@jonathanlew 	@akhilveeraghanta @garethellis0 thoughts?

jonathanlew, 2020-08-07:
Maybe as a safety thing to know if a robot is charged?

MathewMacDougall, 2020-08-07:
Possibly. Now that you mention it, it was very lightly used for that at previous RoboCups. IIRC there were cases during gameplay where when someone went to pick up a robot (to sub it or something) they'd cover it so it would go out of vision and discharge. If we didn't hear it discharge, we'd check the software / "charged light" before picking it up or taking appropriate measures.

IMO I'd rather not put a "charge bar" next to each robot on the field drawing because it isn't relevant gameplay information. Perhaps we make a "status widget" that displays a very small subset of important information we're in communication with. RJ has something sort of like that. This is where #1535 would be more useful to free up more space.

jonathanlew, 2020-08-07:
> IMO I'd rather not put a "charge bar" next to each robot on the field drawing because it isn't relevant gameplay information. Perhaps we make a "status widget" that displays a very small subset of [...]

Agreed

garethellis0, 2020-08-08:
Yes, something like a little icon on each robot on the sidebar (if that's how we're going to display robot info) that lights up if the capacitors are charged would be very helpful. From an architecture perspective I think that we should definitely send back the exact voltage from the robot, and let the visualizer decide on the best way to display it.

jonathanlew, 2020-08-08:
The FullSystemGUI takes SensorMsg directly iirc, so that would contain the voltage

jonathanlew, 2022-01-18:
rescoping into thunderscope project

closed 2022-01-18 by jonathanlew