UBC-Thunderbots/Software #1510: Setup Virtual Network Interface

akhilveeraghanta, 2020-06-22:
### Description of the task

All computers have different interfaces, currently we have the network interface defined as a dynamic param. Ideally we should have a uniform way of using the network to avoid passing in too many arguments
### Acceptance criteria

- [ ] Figure out if this is possible
- [ ] Setup a vitual interface and add it to setup_software.sh
- [ ] Test

### Blocked By
N/A

jonathanlew, 2020-06-22:
Easiest way would be to have the user specify their interface as an argument to `setup_software.sh`, but autodetection would be sick

akhilveeraghanta, 2020-07-06:
This is not super easy to do, there is no clean way to achieve this

We asked on SO and it seems like passing in the argument is the best way to go: https://stackoverflow.com/questions/62744982/how-to-make-a-program-agnostic-of-the-network-interface

jonathanlew, 2020-07-08:
Closing because #1585 largely addresses the issue

closed 2020-07-08 by jonathanlew