UBC-Thunderbots/Software #1534: Visualizer: Allow AI control tab and logs to be hidden so world view can take up more space

MathewMacDougall, 2020-06-30:
### Description of the task

As titled

### Acceptance criteria

- [ ] As titled

### Blocked By

MathewMacDougall, 2020-08-07:
So this is either a trivial ticket that I can do super easily, or could be more complicated depending on what people want for the UI/UX. Qt allows split-panes (which we use to separate the control tabs, logs, and visualization) to be "collapsible", meaning they will disappear if they are dragged below their minimum size. They can be re-opened by dragging them back out.

The only issue with that is that if a pane is hidden/collapsed it's not super obvious something is being hidden. This is likely not an issue if the default state of the GUI is to have all panes open, since the user will then know if they collapsed something. If we want a more "visually obvious" way of collapsing and re-opening panes (for example a "minimize" button like some applications have) this would be a bit more complicated and require some investigation as I don't know how to do that off the top of my head.

Polling @akhilveeraghanta @jonathanlew @kelvinkoon @jared-paul @LiCody @HaomiaoZ for if you think the "easy" way is sufficient or if we want to investigate an alternative solution.

MathewMacDougall, 2020-08-07:
[An example of a split pane "collapsing"](https://youtu.be/fR57o_asTs0?t=97) (ours will be a bit more obvious because there's a visual difference between the contents of each side)

HaomiaoZ, 2020-08-08:
Is it possible to assign a shortcut key to open or close a specific panel and would that be easier than a minimize button?

MathewMacDougall, 2020-08-10:
Looks like there's "hide" and "show" functions, so shouldn't be too bad to set up shortcut keys as long as we can determine what child elements are what widgets. This code would likely be 99% the same for a minimize button (it would trigger basically the same logic I'd imagine).

I figured the main issue with the drag-collapsing was the lack of immediate "intuitiveness" from the user POV, and I imagine the argument would be the same for keyboard shortcuts (which are great for power uses but typically require a more intuitive way alongside them)

jonathanlew, 2020-11-18:
resolved by https://github.com/UBC-Thunderbots/Software/pull/1797

closed 2020-11-18 by jonathanlew