UBC-Thunderbots/Software #1535: Visualizer: Draw field vertically

MathewMacDougall, 2020-06-30:
### Description of the task

It will take up less space overall

### Acceptance criteria

- [ ] As titled

### Blocked By

MathewMacDougall, 2020-08-07:
Polling @akhilveeraghanta @jonathanlew @kelvinkoon @jared-paul @LiCody @HaomiaoZ for if you think this is a good/worthwhile change or not.

HaomiaoZ, 2020-08-08:
So when the field is taking less space means we have more space for? Personally I prefer a horizontal field since it feels more natural to me, but I'm total ok with changing it to vertical if space is an issue now

MathewMacDougall, 2020-08-10:
My initial reasoning was that we don't need the entire width of the screen for logs, since most of those messages are pretty short. Making the field vertical would in theory offer more space for things like robot status (eg. batter and capacitor voltage). I was initially inspired by the Robojackets UI.

I could always make a quick mockup and we can use that to help decide what may be better. For example, we could leave the field horizontal and just put the status information where the logs are.

MathewMacDougall, 2020-09-09:
[image]

Quick mock GUI with a vertical field. It's pretty close to "to-scale" on a 16:9 ratio screen. I assumed we'd show the entire field, which is 9.6/6.6 = about a 1.45 ratio.

PR #1797 (2020-11-12): Rearrange FullSystemGUI to Save Space

## Please fill out the following before requesting review on this PR

### Description
* Rearrange FullSystemGUI to Save Space
* Change dynamic parameter widget for doubles to slider
* Add labels for `Config`s

### Testing Done
manual testing

### Resolved Issues
resolves #1535 (This PR saves space in a different way but achieves the same goals)

### Length Justification

### Review Checklist

**_It is the revie
closed 2020-11-12 by jonathanlew