UBC-Thunderbots/Software #1579: Unify Kick And Chip Primitives To Match Firmware

garethellis0, 2020-07-05:
### Description of the task
There are currently `Chip` and `Kick` primitives in `Software` that both map to a `Shoot` primitive in `Firmware`. We should unify these so that there is a 1-to-1 mapping between software and firmware primitives.

### Acceptance criteria

- [ ] unify `kick` and `chip` primitives in `software` into a single primitive
- [ ] rename the unified software primitive and the firmware `shoot` primitive to have the same name (exact name TBD)

### Blocked By

jonathanlew, 2020-07-17:
@garethellis0 I think that firmware should split shoot into chip and kick, because the parameters are semantically different in the chip and kick case. 
```
    if (chip)
    {
        app_chicker_enableAutochip(chicker, shoot_power);
    }
    else
    {
        app_chicker_enableAutokick(chicker, shoot_power);
    }
```
In this snippet we see `shoot_power` being used to represent `distance_m` in the call to `void app_chicker_enableAutochip(Chicker_t* chicker, float distance_m)`, as well as `speed_m_per_s` when calling `void app_chicker_enableAutokick(Chicker_t* chicker, float speed_m_per_s)`.

garethellis0, 2020-07-17:
Totally fine with me.

jonathanlew, 2020-07-17:
I can reuse this [idea](https://github.com/UBC-Thunderbots/Software/issues/1415#issuecomment-659845328), so I will do https://github.com/UBC-Thunderbots/Software/issues/1577 first

jonathanlew, 2020-07-18:
@akhilveeraghanta made a great point about kick being able to use the geneva drive and chip not being able to, so the plan is to split the shoot_primitive into kick and chip

garethellis0, 2020-07-21:
I thought we agreed to split `shoot` into `kick` and `chip` above?

jonathanlew, 2020-07-21:
that's right we're going to split it

jonathanlew, 2020-07-31:
superseded by #1632

closed 2020-07-31 by jonathanlew