UBC-Thunderbots/Software #1600: Add Checks For Merge Conflict Markers In Formatting Scripts

garethellis0, 2020-07-10:
### Description of the task
We can use `git diff --check` to check for merge conflict markers in files. This should be a blocking check in CI. This could be a check in CI or maybe part of the `fix_formatting.sh` script. Adding it to `fix_formatting.sh` seems a little deceptive since it's a *check* not a *fix*, but at the same time we probably want to be able to run this check locally, not just in CI. 

### Acceptance criteria

### Blocked By

akhilveeraghanta, 2020-07-10:
This is really good, can we also automatically add newlines at the end of the file, saves a lot of "add newline at EOF" comments

https://unix.stackexchange.com/questions/31947/how-to-add-a-newline-to-the-end-of-a-file

jonathanlew, 2020-07-14:
For this to run correctly locally, you would need to git fetch upstream first, right?

garethellis0, 2020-07-17:
> For this to run correctly locally, you would need to git fetch upstream first, right?

I don't _think_ you would have to?

We definitely do not want to start doing push/pulls for people.

PR #1775 (2020-10-17): Added new line EOF and merge conflict markers to fix_formatting.sh

## Please fill out the following before requesting review on this PR

### Description
* Added new line EOF and merge conflict markers to fix_formatting.sh
* improved readability of output
* added new lines to files missing them

### Testing Done
manual testing with script

### Resolved Issues
resolves #1600

### Length Justification

### Review Checklist

**_It is the reviewers resp
closed 2020-10-17 by jonathanlew