UBC-Thunderbots/Software #1607: Investigate Gamecontroller usage in Simulated Tests

MathewMacDougall, 2020-07-16:
### Description of the task

SSL Gamecontroller has a CI mode that we could use to handle all the validation of the ssl rules as opposed to writing validation functions ourselves (this may also require running a refbox).

Investigate to see how easy it would be to use and if the complexity of integration is worth it.

Pros
* Use the most up-to-date rules as defined by the SSL. No bugs from us mis-writing rules
* Easy to update for rule changes by downloading a new binary
* We don't have to write any functions for SSL rules

Cons
* Possibly more complexity to integrate
* Possibly harder to use to debug if messages and failure modes aren't clear

### Acceptance criteria

- [ ] See if it's worth using the gamecontroller in CI mode

### Blocked By

jonathanlew, 2020-07-30:
@MathewMacDougall Were there any more potential issues that you had in mind going into this investigation?

MathewMacDougall, 2020-07-31:
Yea a few other things to consider:
* The Gamecontroller itself isn't what makes rule decisions. It's more of a manager / interface. To my knowledge, the `refbox` is what actually makes those decisions and communicates with the gamecontroller. Eg. [Mannheim Tigers maintains their own implementation of the refbox](https://github.com/TIGERs-Mannheim/AutoReferee). So in order for this to work, we may need to integrate / use **_both_** the gamecontroller and at least 1 autoref. This may add additional complexity, but there should be some examples of integration in the Mannheim or ER-Force repositories for their refbox

akhilveeraghanta, 2021-01-26:
Closing for lack of activity, can re-open if there are any updates

Hi Akhil,

As you already noticed, there are some mechanisms for the integration. We actually do exactly what you described: We have an internal simulator that can be paused and that can run in non-real time. It is fully integrated with the GC and our autoRef.
That's why the GC has the CI mode.
The main use case of the CI mode is that you can control the time. The GC will get the timestamp from your software and will only do something when you send a CiInput. And all it does will be sent back in the CiOutput. So you will have a more reproducible behavior, you can pause the game and run in non-real time.
Since one of the most recent versions, the referee messages will be published if a publish address is set, so that you can consume it in external software as well.
I've experienced some issues with multicast in our pipeline when running multiple builds, so I've avoided all multicast traffic.
But if you want, you can use the vision mode, which takes the time from the vision messages. In the default "system" mode, the time is taken from the operating system.

I've actually never thought too much about integrating the standalone autoRef into this setup. It is running within our AI software for us.

The TIGERs autoRef would probably not run well when the time is not real time, because it synchronizes the vision timestamps to an internal clock that is running in real-time.

So, here some thoughts:
* I'll talk with ER-Force, if they have a solution
* I'll check if the clock synchronization may be adapted
* I could imagine a TCP interfaces for vision and referee messages in the autoRef, if multicast is an issue for you as well

Regarding your setup in general, one way would be to use docker-compose to startup all the components. You can also add the vision-client and status-board for debug purposes:

  ssl-vision-client:
    image: robocupssl/ssl-vision-client:latest
    command: -visionAddress 224.5.23.2:11006 -visualizationAddress 224.5.23.2:11011
    ports:
      - "8082:8082"

  ssl-status-board:
    image: robocupssl/ssl-status-board:latest
    command: -refereeAddress 224.5.23.1:11003
    ports:
      - "8084:8082"

The GC is best integrated into your software directly, so that it can also be used during manual testing. ER-Force is also working on a unix socket integration instead of TCP connections to avoid port conflicts when running multiple instances, but I do not know the progress.

Regards,
Nicolai

On Wed, 26 Aug 2020 at 16:48, Akhil Veeraghanta <akhil.veeraghanta@outlook.com> wrote:

    Hey Nicolai,

    I am on the software team at UBC Thunderbots, and Im currently trying to setup the game-controller as part of our CI loop to verify that our AI doesn't break rules.

    We built up our own custom simulator that allows us to do this, and we are currently in the phase of integrating w/ the game-controller.
    I just wanted to reach out to you with a few questions to make sure we are using the game-controller correctly.

    1. In CI Mode, all multicasting is disabled. We setup the TCP connection w/ the CI input and CI output proto msgs. Our simulator sends the CI Input based on the state of the world. But since multicasting is disabled, we can’t run an AutoRef to verify that our AI isn’t breaking rules like ball speed exceeding max, etc. How can we run the AutoRef in CI mode to test our AI? Or is CI mode to test AutoRefs?

    2. Does the game-controller have its own concept of time, or does it only rely on the timestamps from the vision packets? We could run the game-controller in “vision” mode, and have our simulator output vision packets, run the Autoref, and then have the validation system how we like. However, we don’t run the simulator in “real time” when we are running tests. We pause simulation, update variables, update physics and all that, before looping around to the next step. So we would be updating the timestamps in consecutive vision packets much slower than the actual elapsed time.

    3. Generally how would you recommend integrating game-controller to test that our AI is not breaking rules?

    Thank you for your time!
    Akhil

closed 2021-01-26 by akhilveeraghanta