UBC-Thunderbots/Software #1622: Log Errors From Trajectory Planner In Move Primitive

garethellis0, 2020-07-18:
### Description of the task
The trajectory planner returns error codes to tell us if and how trajectory generation failed. We should log these errors and send them back to software so they're visible to the user. Note that there may not yet be a logging system in place for logging errors from firmware. Please consult with senior software folks to check this!

### Acceptance criteria

- [ ] Log errors returned by the trajectory planner in the move primitive in firmware

### Blocked By

MathewMacDougall, 2020-07-18:
Since the firmware is running a trajectory sent to it by software, do we even need to send errors back? Wouldn't we encounter the error in software first if one exists, so can log it from there?

garethellis0, 2020-07-21:
In an ideal world yes, but the AI is making decisions based on the last perceived state of the robot, and the robot state will change by the time we're actually executing the trajectory on the robot.

MathewMacDougall, 2020-07-21:
Fair enough, and a system to log onboard errors would be more generally useful. Seems like this is something we could add to RobotStatus easily since the trajectory errors are basically enums anywway.

closed 2022-01-18 by akhilveeraghanta