UBC-Thunderbots/Software #1629: Add `static const std::string NAME` to every Tactic and return that value in the `getName()` function

jonathanlew, 2020-07-21:
### Description of the task
We want to have access to the name of a type of `Tactic` in a static way, e.g. `MoveTactic::NAME` to access the return value of the `getName()` function, without instantiating a `MoveTactic`. We want this to be consistent with the Play class.

### Acceptance criteria

- [ ] Add `static const std::string NAME` to each *_tactic.h
- [ ] Set the value of name in the .cpp like so: `const std::string MoveTactic::name = "Move Tactic";` in each *_tactic.cpp
- [ ] return this value in the getName() function

### Blocked By

MathewMacDougall, 2020-07-22:
Just curious if there is a use-case beyond consistency? Are we arbitrarily creating tactics just to get the name somewhere?

jonathanlew, 2020-07-22:
Yeah, basically for consistency.

PR #1674 (2020-08-18): Replace getName and ::name with TYPENAME macro

## Please fill out the following before requesting review on this PR

### Description
Replaced `getName()` and `::name` functions with `TYPENAME` macro. This saves `Tactic`, `Play` and `Backend` from defining `::name` and `getName`. Note that `GenericFactory` will use this too.

### Testing Done
unit tests

### Resolved Issues
resolves #1629 
resolves #159

### Length Justification

### Review Checklist
closed 2020-08-18 by jonathanlew