UBC-Thunderbots/Software #1631: Rename tick and end functions to app_[primitive_name]_primitive_[tick/end]

jonathanlew, 2020-07-22:
### Description of the task
For example, rename `spinning_move_tick` to  app_spinning_move_primitive_tick

### Acceptance criteria

- [ ] Rename all such instances in firwmare/app/primitives folder

### Blocked By

Resocram, 2020-09-28:
@jonathanlew Just wondering, should the format be app [primitive_name] primitive tick or app primitive [primitive_name] tick? This [code style guide](https://github.com/UBC-Thunderbots/Software/blob/master/docs/code-style-guide.md) says it should be PREFIX_FILENAME_NAME, so would it be app primitive [primitive_name] tick?

jonathanlew, 2020-09-28:
I think an example might help. Taking the example of spinning_move_primitive, the filename is `spinning_move_primitive.c`, so the prefix(app or io) is `app`, the filename is `spinning_move_primitive` and the name is `tick`, then the full name would be ` app_spinning_move_primitive_tick`, right?

Resocram, 2020-09-28:
Oh sorry I was looking at the directory containing those files and it was named primitive, so I got confused. Thanks for the clarification!

PR #1756 (2020-10-04): Renamed tick functions to app_[primitive_name]_primitive_tick

## Please fill out the following before requesting review on this PR

### Description

Renamed all the tick functions located in the firmware/app/primitive folder to follow app_[primitive_name]_primitive_tick
### Testing Done

### Resolved Issues

Resolves #1631 

### Length Justification

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has b
closed 2020-10-04 by Resocram