UBC-Thunderbots/Software #1639: Create PrimitiveMsg Factory

jonathanlew, 2020-07-28:
### Description of the task
Create primitive_msg_factory[.h/.cpp] in software/primitive for creating `PrimitiveNewMsg`s. The file will contain functions like
`std::unique_ptr<PrimitiveMsg> createAutokickMovePrimitive(const Point &dest, const Angle &final_angle, double final_speed, double dribbler_speed, double kick_speed)` for all `*PrimitiveMsg` in primitive.proto except for `DirectControlPrimitiveMsg`

NOTE: while this will be called PrimitiveMsgFactory, it will create PrimitiveNewMsgs as defined in primitive.proto since as a step before integration for the new primitive protos. During integration, we will rename PrimitiveNewMsg to PrimitiveMsg

### Acceptance criteria

- [ ] New factory functions
- [ ] Tests

### Blocked By
#1620

MathewMacDougall, 2020-07-29:
Curious why this factory takes the primitive parameters directly, rather than a specific Primitive class?

eg. why not `std::unique_ptr<PrimitiveMsg> createPrimitive(const MovePrimitive& primitive);`?

jonathanlew, 2020-07-29:
With the new primitive protobufs being more expressive, we can do away with the Primitive class entirely, so this issue it targetting that new architecture. https://github.com/UBC-Thunderbots/Software/pull/1637 is the first step in that direction

MathewMacDougall, 2020-07-30:
Ah ok if the current Primitive class doesn't exist / is removed this makes way more sense :+1:

PR #1646 (2020-08-01): Create PrimitiveMsg Factory

## Please fill out the following before requesting review on this PR

### Description

Create `primitive_msg_factory` to create messages for:
- `ChipPrimitiveMsg`
- `KickPrimitiveMsg`
- `MovePrimitiveMsg`
- `SpinningMovePrimitiveMsg`
- `AutochipMovePrimitiveMsg`
- `AutokickMovePrimitiveMsg`
- `StopPrimitiveMsg`

### Testing Done

Tests in `primitive_msg_factory_test.cpp`

### Resolved Issues

Resolves #1639 

### Length Justification

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been covered_**

- [ ] **Function  [... cut]
closed 2020-08-01 by kelvinkoon