UBC-Thunderbots/Software #1649: Create controller abstraction that takes RPM around app_wheel_applyForce

jonathanlew, 2020-07-30:
### Description of the task
We should have a a controller abstraction that wraps applying force to the wheels and attempts to achieve the given rpm. 

### Acceptance criteria

- [ ] Controller abstraction
- [ ] Testing with robot

### Blocked By

garethellis0, 2020-08-01:
Please note that modifications to the controller on the old robots are very dangerous and you can easily damage/destroy mainboards if you are not careful.

jonathanlew, 2020-08-01:
iirc the old controller is bang bang on force. Is it easy to map rpm to force?

MathewMacDougall, 2020-08-03:
@jonathanlew the old controller for robot position was bang-bang on jerk (not force)

I'm not sure what the control scheme is at the wheel-level. Looking it up, applying force calculates a PWM value to send to the wheel based on current voltages. These PWM values get stores in a static variable which I'm guessing is then send over ICB to the FPGA to handle the commutation.

So for the wheels there's no real "control scheme" when we applyForce, so getting an "accurate" mapping may be really tricky.

jonathanlew, 2020-08-03:
hm, if trajectory planner calculates wheel speeds, how can it interface with this system?

MathewMacDougall, 2020-08-05:
you could potentially try some mapping of `speed -> force` but it would be tricky to get an accurate model. And that wouldn't include a separate controller to get the wheel to a certain speed/force quickly.

MathewMacDougall, 2020-08-05:
Depending on how much you want to "properly" support the old robots, it may be something to just try manually create and tweak a mapping from speed to force and iterate on the field until it's "good enough".

jonathanlew, 2020-08-05:
@garethellis0 Wondering what you think about this with trajectory planner integration given the trajectory planner outputs a target velocity that gets converted into target wheel speeds

jonathanlew, 2020-08-05:
imo "good enough" is probably good enough. A key part of the issue here is whether `app_wheel_applyForce` is a good app layer interface and whether this should be `app_wheel_setTargetRPM` or something along those lines.

MathewMacDougall, 2020-08-06:
Assuming we don't want to support both simultaneously, I think `app_wheel_setTargetRPM` will be a better app-level interface for the trajectory planner and new control. Then we can just change the injected function for the old robot firmware.

garethellis0, 2020-08-08:
Based on current discussions around how the per-wheel control is going, there is a serious possibility that if we go with the trinamic motor controllers then we will just be directly setting the wheel velocity. If we're using the Allegro ones then there will be some more firmware in IO. 

Based on the above, my recommendation would be to have the app layer encompass the common parts to both the above proposals, ie. setting wheel speeds. 

TL;DR: I agree with @MathewMacDougall

PR #2062 (2021-05-10): Implemented Velocity Wheel Trajectory Follower

## Please fill out the following before requesting review on this PR

### Description
* Implemented Velocity Wheel Trajectory Follower
* Adjusted the interfaces slightly to add num_elements
* Added setTargetRPM for clarity

### Testing Done
pending

### Resolved Issues
resolves #1649 
resolves #2044

### Length Justification

### Review Checklist

**_It is the reviewers responsibility to also make sur