UBC-Thunderbots/Software #1650: Clarify which firmware function comments require [in/out] specifiers

jonathanlew, 2020-07-31:
### Description of the task
A lot of firmware does not have [in] specifiers when it is clear that they are inputs. For example in `double calculateSomething(double arg1, double arg2, double arg3)`, it is clear that the args are all inputs since they are primitive types. This issue is about clarifying how this works, documenting it and updating the code where it is inconsistent.

### Acceptance criteria

- [ ] documentation
- [ ] update code to match new convention

### Blocked By

MathewMacDougall, 2020-07-31:
> it is clear that the args are all inputs since they are primitive types

Playing devil's advocate, simply because an arg is a primitive type does not mean it can't be / isn't set by reference. Eg. there may be cases (not claiming the exist in our code atm) where we return an error code, so need to set a param by value.

jonathanlew, 2020-07-31:
I don't think you can pass by reference in C, so specifically I mean primitive _non-pointer_ arguments

MathewMacDougall, 2020-08-03:
Yea sorry I meant pointer above. Primitive non-pointer args seem like a relatively specific subset for people to watch out for, so it may be easier to just say everything requires the in/out.

The trajectory planner is currently the most complex code that makes heavy use of mutable pointer-args (mostly non-primitives though). On the other hand there's libs like physics and util stuff that pretty much only takes non-pointer primitive args. So it seems like places were we do / don't need the specifiers is somewhat localized, although I don't think that's a great argument for either case.

jonathanlew, 2020-08-07:
I'm in favour of only pointer args requiring [in/out] specifiers

MathewMacDougall, 2020-08-07:
I'm fine with that. At least a clear split of what needs it or not, and pointers are the only things that need clarifying anyway.

PR #1706 (2020-09-04): Standardize Firmware Javadoc [in/out] Specifiers

## Please fill out the following before requesting review on this PR

### Description

Update coding style docs for how pointer arguments need [in/out] specifiers. Fix firmware function Javadoc comments.

### Testing Done

N/A. Simply fixing comments.

### Resolved Issues

Resolves #1650 

### Length Justification

### Review Checklist

**_It is the reviewers responsibility to also make sure every item here has been covered_**

- [ ] **Function & Class comments**: All function definitions (usually in the `.h` file) should have a javadoc style c [... cut]
closed 2020-09-04 by kelvinkoon