UBC-Thunderbots/Software #1651: Change BallCollisionType to bool

jonathanlew, 2020-07-31:
### Description of the task
bool might be clearer than an enum: https://github.com/UBC-Thunderbots/Software/pull/1637#discussion_r463419466

### Acceptance criteria

- [ ] Change BallCollisionType to bool 

### Blocked By

MathewMacDougall, 2020-07-31:
@garethellis0 I know you have opinions about this

Pros of bool
* Can be clearer if there is only a single bool argument, and there are truly only 2 states for something (ie. the bool isn't conditional logic for some other parameter like we used to have with kick/chip)
* slighlty simpler to handle (no case statement)

Cons of bool
* Arguable less readable at a glance if the purpose isn't somehow super obvious based on the function

Eg. `MoveToDestination(true)` is less readable than `MoveToDestination(BallCollisionType::AVOID)` because in this context it's not immediately super obvious what that bool could mean.

So possibly worth leaving it as an enum since we have it already, but your guys' call.

jonathanlew, 2020-07-31:
@jordanzliu

garethellis0, 2020-08-08:
After thinking about this a bit, I agree that the meaning of the boolean should be clear from the site of the function call. So in this case an enum would probably be more appropriate. But if we had a function like `setAvoidBall` I think it makes sense for that to take a boolean. Just my two cents.

akhilveeraghanta, 2020-09-18:
@EvanMorcom has made efforts to turn other booleans into enums, so this ticket is currently  invalid as it would also involve reverting those changes -> https://github.com/UBC-Thunderbots/Software/pull/964

closed 2020-09-18 by akhilveeraghanta