UBC-Thunderbots/Software #1652: Create Firmware Logger

jonathanlew, 2020-08-01:
### Description of the task
Figure out a way to log errors in firmware and send it for display in software

### Acceptance criteria

- [ ] special protobuf message
- [ ] multicast listeners/senders
- [ ] display with g3log in software?

### Blocked By

MathewMacDougall, 2020-08-03:
Do you see this replacing the current ErrorCode system or working with it / building on top of it?

jonathanlew, 2020-08-03:
@akhilveeraghanta and I talked about this offline a while back. ErrorCode should be for hardware-related or system crashes, while the firmware logger should be for warnings and debugging. The firmware logger can accumulate messages and send it up in batches, while the ErrorCode should be sent as fast as possible to alert AI

closed 2020-11-11 by akhilveeraghanta