UBC-Thunderbots/Software #1660: Remove `end` from firmware primitives

jonathanlew, 2020-08-08:
### Description of the task
The `end` function in firmware primitives is redundant because `app_primitive_manager_makeRobotSafe` accomplishes anything that the `end` function could or better. 

### Acceptance criteria

- [ ] Remove `end` from firmware primitives

### Blocked By

MathewMacDougall, 2020-08-10:
`makeRobotSafe` discharges the robot, which we may not always want when switching primitives?

jonathanlew, 2020-08-11:
`app_primitive_manager_makeRobotSafe` is called every time a primitive ends and it does not discharge the robots

MathewMacDougall, 2020-08-11:
Ah. I may have missed this whenever we had discussion about what `app_primitive_manager_makeRobotSafe` should do, but I'd assume "safe" means discharging the robots as well so that they can be handled by a human. I know I also have been vocal _against_ discharging caps between primitives, so this doesn't totally align.

IMO `makeRobotSafe` _should_ discharge the caps as well, as we should either have a slightly different function that gets called when primitives end, or _maybe_ we add a parameter to determine if we discharge or not when making the robot safe.

Eg. we could either have
`makeRobotSafe(enum Discharge)`

or

```
makeRobotSafe() {
    stopRobot();
    discharge();
}
```
and call `stopRobot()` at the end of primitives

PR #1741 (2020-10-10): Added Estop Primitive and Fixed Charge/Discharge Interface

## Please fill out the following before requesting review on this PR

### Description
Added Estop Primitive and Fixed Charge/Discharge Interface
* charging capacitors moved from old firmware's receive.c to app layer's primitive_manager.c
* Added estop primitive
* refactored makerobotssafe and stop robot

### Testing Done
might be able to test in sim

### Resolved Issues
resolves #1679
resolves #1660
resolves #1728

### Length Justification

### Review Checklist
closed 2020-10-10 by jonathanlew