UBC-Thunderbots/Software #1666: Add networking setup to documentation to run Simulator

MathewMacDougall, 2020-08-11:
### Description of the task

While trying to run the Simulator with 2 AIs, we realized by Ubuntu closes all ports by default (no data goes through), so we need to explicitly open the ports we use to communicate in order for things to work.

Eg.
```
sudo ufw allow 42069
sudo ufw allow 42070
sudo ufw allow 42071
```

We should figure out how we want to involve this in our environment setup. Options:
* Add it to the documentation
* Add it to the setup scripts
* Something else?

In all cases it's a bit tricky since we'd have to make sure the values are up to date with our internal constants, or tell the user to check the contants file for the most up to date values. Thoughts on the best way to do this?

### Acceptance criteria

- [ ] Figure out how to set up networking ports in our environment setup
- [ ] Do it

### Blocked By

MathewMacDougall, 2020-08-11:
@akhilveeraghanta @jonathanlew @kelvinkoon @jared-paul @LiCody @HaomiaoZ

kelvinkoon, 2020-08-11:
I think you can do `sudo ufw allow 42069:42071` to keep it a one-liner.
If you're doing a script, maybe also do `sudo ufw status` to see that it worked.

My intuition about the ports is they shouldn't change much so I don't see anything wrong with telling them to check the constants file and the documentation pointing to it.

kelvinkoon, 2021-01-25:
Confirmed by @jsn0716 no additional setup was needed with port configurations/ufw. I think maybe just adding a note that if `ai vs ai` isn't running, try this series of commands.

PR #1916 (2021-02-13): Add Networking Port Support for AI vs AI
## Please fill out the following before requesting review on this PR

### Description

Quick changes to open ports 42069 to 42073 for `ai vs ai`. 
In testing, most people were able to get `ai vs ai` running but in the odd case, the ports need to be explicitly opened.
Documentation in the `Getting Started` section was also updated.

### Testing Done

Running the Bash script gave a status that was expected.

### Resolved Issues

Resolves #1666 

### Length Justification

### Review Checklist

**_It is the reviewers responsibility to also make sure every i [... cut]
closed 2021-02-13 by kelvinkoon