UBC-Thunderbots/Software #1712: Investigate Warnings from SimulatedTests

jonathanlew, 2020-09-05:
### Description of the task
Several SimulatedTests cause warnings that fill the logs and are highly suspicious and may indicate a bug. One example is `free_kick_play_test`, which causes `Could not draw a line from the goalie to the ball through the defender crease` and `Error updating robot state, stopping` warnings. To start this investigation, run `//software/simulated_tests:free_kick_play_test` in debug mode in gdb with breakpoints where those warnings are logged. 

### Acceptance criteria

- [ ] Explain why we get so many warnings from simulated tests
- [ ] Figure out the best way to avoid these warnings, e.g. fixing an underlying bug or suppressing warnings that aren't helpful

### Blocked By

amrm3lm, 2020-10-24:
I am still looking into the warning caused by the navigator, but I thought Id give an update. When I ran the tests, I got the following 3 warnings:

**WARNING [team.cpp->assignGoalie:108]	Warning: Tried to assign the goalie (id 0) to a robot that is not a member of the team**

Seen in: example_play_test

I decided to ignore this warning for this play because the example_play is deliberately setup to not have a goalkeeper, and the simulated test acknowledges that in a comment. Suppressing it is an option too. 

----------------------------------

**WARNING [crease_defender_tactic.cpp->calculateDesiredState:114]	Could not draw a line from the goalie to the ball through the defender crease**

Seen in: freekick_play_test, defense_play_test, enemy_free_kick_test, shoot_or_chip_test, shoot_or_pass_test, basically anything that uses the crease_defender_tactic.

2 main reasons we are seeing this warning:

1. The temporary setup of simulated tests that dont have validation functions yet didnt have goalkeepers in the defense area, which is required to run the crease defender tactic. This is an easy fix. 
2. Comparisons of doubles leading to inaccurate results.

In our `contains::contains(ray, point)`, we try to find whether the point is collinear with the ray. We use FIXED_EPSILON = 1e-15 and use the defualt ulps_10. The result is that `almostequals()` in `collinear::collinear(point a, point b, point c, FIXED_EPSILON, ulps_10)`  isnt returning the value we expect it too. I tried using `FIXED_EPSILON=1e-10` and that removed all warnings. Similarly, using `ulps_100` solves the issue. Most comparisons that cause the warning had a ulps distance of 75.

I think passing the value of EPSILON/ulps that we want to use into the contains method would solve the problem easily (that is if we agree that `1e-10` or `ulps_100` is a good enough precision).

Also in the `contains` method we find whether a point is in the ray direction the following way. This was returning false when it should have been returning tru, directly leading to the warning:

`bool point_is_in_ray_direction =
        ((contained - container.getStart()).normalize() - container.toUnitVector())
                .length() < FIXED_EPSILON;`

I think a better way would be to avoid comparisons around 0, so something like this:
`bool point_is_in_ray_direction = almostEqual(((contained - container.getStart()).orientation().toDegrees()),container.toUnitVector().orientation().toDegrees(), FIXED_EPSILON, ULPS_100);`

I will also note that there are lots of calls from other methods where the difference between two values is less than 1e-10, but almostequals() returns false, so it might be worth looking at those cases too. The crease_defender_tactic is a bit different from these cases in that we know with certainity that there exists an intersection between the goalie_to_ball ray and the friendly_defense_area, and so all the calls to almostequals() which are used to verify this intersection should return true.

---------------------------------------------------------------------------------------------------------------------------------------------------------------------------

**WARNING [navigator.cpp->getAssignedPrimitives:69]	Navigator's path manager could not find a path for RobotId = 5**

Seen in: freekick_play_test, defense_play_test, enemy_free_kick_test, shoot_or_chip_test, shoot_or_pass_test, kickoff_friendly_play_test, penalty_kick_enemy, penalty_kick. Lots of occurrences. Not limited to RobotId=5. 

So far I think the main issue is that we are assigning pathObjectives but then have motion constraints like "FRIENDLY_DEFENSE_AREA" that make getting to the objective impossible. I am still learning about the navigator though so ill have a better picture of whats going on soon.

jonathanlew, 2020-10-25:
Good work. 

I agree with ignoring `example_play_test`, because it's such an exceptional situation

For `WARNING [crease_defender_tactic.cpp->calculateDesiredState:114] Could not draw a line from the goalie to the ball through the defender crease`, it shouldn't be too hard to add goalies to the crease and increasing the margins for almostequal seems reasonable. I think it would be good to get these changes in a PR. 

`WARNING [navigator.cpp->getAssignedPrimitives:69] Navigator's path manager could not find a path for RobotId = 5` It would be good to narrow down the specific situation that it happens to figure out what's happening.

PR #1849 (2020-12-04): Fixed intersection(Ray, Segment) floating point bug

## Please fill out the following before requesting review on this PR

### Description
* Fixes a bug where a `intersection` between a Ray and a Segment that do intersect returned false
* The root cause was in `contains` and `collinear`
* This bug was discovered as part of #1712 and reduces warnings from the crease defender

### Testing Done
regression tests for all affected functions, manually verified that warnings are gone

### Resolved Issues
resolves #1712

### Length Justification

### Review Checklist
closed 2020-12-04 by jonathanlew