UBC-Thunderbots/Software #1717: RobotFilter does not remove expired robots if there are no new detections

MathewMacDougall, 2020-09-09:
### Description of the task

The BallFilter does not properly remove robots if `getFilteredData` is called with an empty vector.

This can and should be reproduced with a unit test before fixing to make sure it does indeed get fixed and to prevent future regressions.

File: `software/sensor_fusion/filter/robot_filter.cpp`

In `getFilteredData`, if the function is called with an empty vector the first loop does not run. Therefore when we get to the `data_num == 0` check, we still have the temporary `latest_timestamp` `Timestamp` that was created above. Because this value is 0, it will never be greater than the expiry buffer + the latest robot timestamp, so the function will never return `std::nullopt` indicating the robot is no longer detected. Instead it will always return the most recent robot state stored by the class.

This should be fixed so that robots that do not appear in a detection for longer than the `expiry buffer` get reported as "nonexistent" by the `RobotFilter`, even when the function is called with an empty vector repeatedly.

### Acceptance criteria

- [ ] Create a unit test that fails, using the above reproduction
- [ ] Fix the bug so that the RobotFilter will correctly expire robots when `getFilteredData` is called with an empty vector
- [ ] Make sure the unit test passes

### Blocked By

kelvinkoon, 2020-09-24:
As discussed off-line, `robot_filter.cpp` does not keep track of time. It was decided it's worth revisiting to figure out how to refactor to make this ticket more possible. It will be put into `triage` for the time being.

jonathanlew, 2020-11-10:
Probably the best thing for this ticket is to refactor getFilteredData and update the comments with this observation