UBC-Thunderbots/Software #1725: Make physics time step consistent between unit tests and simulation

MathewMacDougall, 2020-09-10:
### Description of the task

Some of our simulation unit tests are designed not just to validate basic functionality, but actually to help us tune certain behavior in simulation.

For example, all the various dribbling tests in `simulator_robot_singleton_test.cpp` are not only designed to make sure dribbling works at all, but to help us tune various values in the implementation and other simulation values. For example, if we define that the robot should be able to rotate at 2rpm before it loses the ball, we can use one of the unit tests to make sure that's the case, and tune force values in the implementation accordingly to get that desired behavior.

However, this means that these tests need to effectively be set up the same as the actual Simulator (standalone and simulated tests). There are values such as
* physics time step
* various restitution values
* various friction values

that can affect the behavior of dribbling, and therefore should be consistent between the tests and "real" simulation. If they are not, then the tests are not as meaningful because a unit test may say we can dribble the ball up to 2rpm, but that may not be the case in the simulated tests.

Currently some of these values are shared because they are `static constexpr` class constants. Most friction and restitution values fall under this category (except for the ball).

However the physics time step is **not** shared between the tests and "real simulation". The time-step has a large impact on how well the robot can keep control of the ball while dribbling, because the dribbler is so small the more chances we have to apply force (even if small), the better control we have.

Eg. The robot can spin with the ball up until about **8 rad/s** with a **1/60s** time step. However the robot can spin with the ball at about **20 rad/s** with a **1/120s** time-step.

Therefore we should find a way to share the physics time-step between the tests and actual simulator (both the standalone sim and simulated tests). Or, find another solution such that we can have tests that help us tune the desired behavior of certain features in the simulation.

### Acceptance criteria

- [ ] Find a solution such that we can have tests that help us tune the desired behavior of certain features in the simulation.
- [ ] Implement said solution

### Blocked By

MathewMacDougall, 2020-09-10:
@jared-paul @akhilveeraghanta @jonathanlew please let me know if this makes sense.

jared-paul, 2020-09-12:
The issue makes sense to me, but my initial thoughts are that it would be very hard to solve

MathewMacDougall, 2020-09-12:
Yea there will for sure be discrepancy between IRL and simulation, but the issue here is discrepancy between the unit tests for simulation code, and the actual simulation (standlone sim or simulated tests).

The main idea that come to mind is to just define the time-step as a public `static constexpr double PHYSICS_STEP` constant in the `Simulator` class. However, this does mean the simulation step size can no longer really be a parameter.

So a new interface may look like

```
// Steps simulation by Simulator::PHYSICS_STEP
void stepSimulation();
```

Alternatively if some users want to control higher-level steps (ie. a camera frame, which may be made up of multiple physics and primitive steps)
```
// Advances the simulation by the closest multiple of Simulator::PHYSICS_STEP
// to the given step. Eg. if PHYSICS_STEP = 0.15 and step=0.35, internally
// the simulator would run 2 physics steps and advance the simulation by
// 0.3
void stepSimulation(Duration step);
```
Since it should always be better to take smaller steps, it may be possible to also take steps `<= PHYSICS STEP` to match the requested step size. However it is also not recommended to take steps of different sizes, but it may be fine here.

MathewMacDougall, 2020-09-12:
Main trick I guess is just making sure the unit tests use the static class variable rather than defining their own values. Helper functions in test fixtures will help with that.

jonathanlew, 2022-03-23:
closed due to new simulator

closed 2022-03-23 by jonathanlew