UBC-Thunderbots/Software #1823: Scope Out Per-Wheel Velocity PID Controller Architecture and Components

jonathanlew, 2020-11-14:
### Description of the task
Scope Out Per-Wheel Velocity PID Controller Architecture and Components:
* integration with trajectory planner
* how to deal with position and orientation error
* translation to per-wheel velocities
* velocity controller
etc.

### Acceptance criteria

- [ ] Architecture doc

### Blocked By

jonathanlew, 2020-11-14:
@garethellis0

garethellis0, 2020-11-20:
OK, so there are roughly three stages here:

## The Absolute Minimum Amount Of Work Required To Have Things Work On The New Robots
Update [this function](https://github.com/UBC-Thunderbots/Software/blob/3dad9e3a4938717198c07b94cf26e1afde150863/src/firmware_new/boards/frankie_v1/io/drivetrain_unit.c#L21-L43) to provide some semblance of an accurate relationship between the force we're trying to apply to each wheel and a pwm value. This is obviously very poor, but is functionally equivalent to what was done on the previous generation of robots. Yes, we did not use the encoders for wheel control :-1:.

## A Wheel Controller With Some Feedback :heart_eyes: 
(1) Design a controller that can be used inside the "drive unit" to set the wheel velocity (`set_wheel_velocity(float radians_per_second)`). A simple PID controller should be fine here. 
(2) Create `app_control_setVelocity(FirmwareRobot, float velocity_x, float velocity_y, float velocity_angular)` that translates the x,y,theta velocities into wheel velocities. *Note:* with 4 wheels, there are multiple sets of 4 wheel speeds that can give you the same x,y,theta. As such, you should consider this as more of a regression problem, and minimize some extra factor (for example, the L2 norm of the wheel speeds) so that you don't get random jumps in wheel speeds.
(3) At every tick in the `move_helper`, figure out what velocity the robot should be at by linearly interpolating between the control points in the trajectory using the current time. Then use the function that you created in (2) to set the robot to that velocity.

This has the problem that for longer trajectories it will accumulate error since you're just controlling on velocity, not position. It may very well be that this is practically not a problem, so we _strongly_ advise that you do everything to get this as good as possible before deciding to move to the next tier (which is _significantly_ harder).

## Bring Some Position Feedback Into Your Control
(1) For every tick in the `move_helper`, choose the next `N` (probably something small, 2-5) control points on the trajectory we're trying to track. For each of these control points, compute a path from the robot's current position to that control point such that the path is tangent to the Robot's direction of motion and also tangent to expected direction of travel at the control point. For each path, compute a trajectory, and select the trajectory that minimizes some quantity (maybe curvature), with the initial velocity being equal to the robots velocity and the final velocity being equal to the desired velocity at the control points. Then just track this new, smaller trajectory.

jonathanlew, 2022-01-18:
closing in favour of trinamics

closed 2022-01-18 by jonathanlew