UBC-Thunderbots/Software #1908: Implement Stop Play Test

jonathanlew, 2021-01-23:
### Description of the task
Write validation functions to ensure that the robots obey the [rules for stopping](https://robocup-ssl.github.io/ssl-rules/sslrules.html#_stop). Add tests for robots close to ball/far from ball + friendlyhalf/enemyhalf 

### Acceptance criteria

- [ ] write validation functions to check rules
- [ ] identify any broken rules and potential solutions

### Blocked By

jonathanlew, 2021-01-25:
We should aim to have a test for robots moving too quickly as discussed here: https://github.com/UBC-Thunderbots/Software/issues/1882

tarakong01, 2021-01-31:
Do you mean as in, test a scenario where robots are moving too fast and make sure the test fails? Or just testing that they are moving <= 1.5 m/s as the rules state?

jonathanlew, 2021-01-31:
The first one. Set up a test scenario where the robots go too fast and a validation function that catches it, then remove that check and add a TODO linked to #1882.

PR #1949 (2021-02-21): Simulated Tests and Validation Functions for Stop Play

## Please fill out the following before requesting review on this PR

### Description

Wrote validation functions and simulated tests for stop play, according to SSL Rules for stopping. Currently commented out all speed checks in tests, linked to #1882 to implement robots slowing down. Last test case is disabled, linked to #1948 to fix bug. Linked other test cases to #1948 as well, for warning messages and bad robot positioning.

Note: will wait on #1946 to merge, before merging this PR (in order to resolve the merge conflicts that will o [... cut]
closed 2021-02-21 by tarakong01