UBC-Thunderbots/Software #1987: Tune PassGenerator Optimizer

akhilveeraghanta, 2021-03-13:
### Description of the task
The PassGenerator currently uses ADAM to optimize for better receive locations and pass speeds.
We put each pass through 10 iterations of gradient descent, and we do that for every pass over 18 zones. However, the pass rating improvement from 4, 10, to 20 to 100 iterations isn't significant enough to justify the time + resources needed.

We are trading off between the speed of finding a good enough pass and finding a _really_ good pass. Since the field is highly dynamic, optimizing on outdated data is always bad.

Here are some plots of how the score improves, more details can be found at the bottom of the page here: https://docs.google.com/document/d/14clpTMeEFrAMnvzEfV9XoediuuFyDeWB4ktByog2EKQ/edit
[image]

10 iterations take ~2 ms each for a total of 18 passes which adds up to (36ms)
2 iterations take 0.4ms each for a total of 18 passes which adds up to (7.2ms)

In the corner kick play, 2 iterations per pass results in a pass w/ a score 0.84 and 10 iterations results in a pass w/ a score of 0.86

So I've temporarily set the max iterations to 2, but we should find the sweet spot by tuning step sizes and weights.

### Acceptance criteria

- [ ] Use our new Jupyter Notebook capabilities and understand how the passes are being optimized
- [ ] Tune the weights we start the optimizer w/
- [ ] Generate data to show that we've improved (this is hard to regression test against, but a histogram of the difference between the current score and the optimized score before and after is indicative of improvement)

### Blocked By
#1986

MathewMacDougall, 2021-06-10:
@akhilveeraghanta were these tests run in `fastbuild` or `optimized`? These results seem really slow compared to what I remeber from the old PassGenerator (but I don't have concrete data).

MathewMacDougall, 2021-06-10:
Actually I ran the cost_function performance test and this doesn't seem too unreasonable, probably different computer play a role. (Still curious though)

jared-paul, 2021-06-10:
> Actually I ran the cost_function performance test and this doesn't seem too unreasonable, probably different computer play a role. (Still curious though)

I think the performance is a lot better now with the new calcbestshot implementation (pretty sure @akhilveeraghanta was able to run 8 iterations safely)

nimazareian, 2024-05-28:
Closing since this is outdated and we're pursuing different pass generator design

closed 2024-05-28 by nimazareian