UBC-Thunderbots/Software #2024: [SSL Simulation Protocol] Add receive function to UDP sender

jsn0716, 2021-04-08:
### Description of the task

Add receiving functional to `ProtoUdpSender`. This is to be able to receive a response from the SSL Simulator after sending a robot control command / simulator command as defined [here](https://github.com/RoboCup-SSL/ssl-simulation-protocol).

### Acceptance criteria

### Blocked By

jonathanlew, 2021-04-08:
How is this different from the listener?

jsn0716, 2021-04-08:
This is to receive a message after sending as a response vs listening to a specified port - see example [here](https://www.boost.org/doc/libs/1_35_0/doc/html/boost_asio/tutorial/tutdaytime4/src.html) where they use the same socket to receive a response after sending a message

jonathanlew, 2021-05-04:
resolved by #2035

closed 2021-05-04 by jonathanlew