UBC-Thunderbots/Software #2048: Critical Path Performance Optimizations (Field, Geom, PathPlanner)

jonathanlew, 2021-05-05:
### Description of the task
Performance improvements found from running `bazel run -c dbg --run_under="valgrind --tool=callgrind --callgrind-out-file=/tmp/profile.callgrind" //software/ai/hl/stp/play:corner_kick_play_test` and looking at the output
Multiply all estimated improvements by 100/65 due to 35% simulator overhead):
* Cache reduced_size_field (cost_function.cpp:315): 1.63%
* Pre-construct defense areas in Field class and return as const& (field.cpp): 1.65%
* Precalculate isUnblocked grid for enemy robots and share among all findPath calls (theta_star_path_planner.cpp): total calls to isUnblocked is 3.05%, so conservatively at least 1.5% can be avoided
* Cache when lineOfSight is blocked by an enemy robot and shared among all findPath calls during a tick (theta_star_path_planner.cpp): total calls to GeomObstacle<Circle/Polygon>::intersects is 12%, so conservatively we can save 5%
* Replace map with unordered map (theta_star_path_planner.cpp): total map insertion and lookup is ~6% but it could be dominated by `computeComparisonKey`, so conservatively estimating ~0.5% 

Total: ~15% (~10%*100/65) conservatively, up to 25% if we're lucky

### Acceptance criteria

- [ ] Make optimization listed above

### Blocked By

nimazareian, 2021-08-21:
If in the future we go back to a design where obstacles are vectored and not "rasterized" at the start of the path planner, some optimizations for checking intersection between path and obstacle could include:
-  Using a KD-tree or similar data structures to only check the obstacles that are close to the line we're checking. This paper goes into detail of various methods used for collision detection that could be applied to a path planner: http://www.cs.cmu.edu/~jbruce/thesis/chapters/thesis-ch03.pdf
- Using a bounding box around polygon and circle obstacles to do a fast initial intersection check, and if required, use more advanced algorithms to check for more precise intersections. This concept is often used in games to check for collisions in 2D and 3D worlds.

closed 2021-08-23 by nimazareian

nimazareian, 2021-09-16:
Possible implementation of contains(polygon) that may be more efficient than current implementation:
https://twitter.com/FreyaHolmer/status/1232826293902888960?s=20
https://github.com/FreyaHolmer/Mathfs/blob/master/Polygon.cs