UBC-Thunderbots/Software #2080: Implement Kicker and Chipper Firmware

akhilveeraghanta, 2021-05-15:
### Description of the task

The power board provides an interface to charger/discharge capacitors so that we can kick and chip. This ticket involves implementing the functions in `charger.c/h` to charge and discharge the capacitors.

`CHARGE_PWR_BRD` pin is the one we should be using, talk to electrical to figure out how we are supposed to toggle that pin.

[image]

### Acceptance criteria

- [ ] Implement charger.c/h
- [ ] Implement the chicker.c/h functios
- [ ] Test!

### Blocked By
The app layer, we need the ability to safely e-stop and discharge before implementing this change.

liambontkes, 2021-06-12:
Breakbeam functionality:

1. Interrupt triggered whenever the Breakbeam is broken.
2. Function to poll the Breakbeam's state (for robot diagnostics).

jonathanlew, 2021-08-19:
out of date given move to power board mcu

closed 2021-08-19 by jonathanlew