UBC-Thunderbots/Software #2088: Fix Floating Point Exception Errors

jonathanlew, 2021-05-17:
### Description of the task
There are floating point exception errors that we ignore, e.g. sqrt of a negative number, overflow, etc. The most visible way this fails is when it causes the robot state to contain `nan`s (not a number), which causes robot costs to be `nan`s, which causes the munkres algorithm to stall. 

Floating point errors can be detected automatically by adding `feenableexcept(FE_INVALID | FE_OVERFLOW);` to the main function or where it has to be called, e.g. in the simulated test fixture constructor. More info:
* https://stackoverflow.com/questions/11933919/debug-c-code-catch-first-nan-appearance
* http://www.cplusplus.com/reference/cfenv/FE_INVALID/

Currently, _every_ simulated test suffers from at least one floating point exception error, typically in physics or primitives functions.

### Acceptance criteria

- [ ] Add `feenableexcept(FE_INVALID | FE_OVERFLOW);` to the simulated test fixture and test of all affected libraries at minimum (as regression tests), but ideally to every gtest
- [ ] Fix all floating point errors encountered by simulated tests

### Blocked By

MathewMacDougall, 2021-05-17:
Ooooh floating point is always really tricky.

For the first error, it's possible the `radius_of_curvature` is very large because the path being navigated is a straight line (where the radius of curvature is technically infinite). IIRC this was thought of and there was some limit to it, but maybe it's not applied everywhere or there's some other way for it to slip through.

**Edit:** There _is_ a check in `firmware/shared/math/polynomial_2d.c` where we check for division by zero and set the result to `FLT_MAX`. I suspect this line is being hit and `FLT_MAX` is being returned, since the value of `FLT_MAX` is very close to what's in the trace above (`1e37` vs `3e38`).

https://www.tutorialspoint.com/c_standard_library/float_h.htm

For the second error given how (relatively) large the `displacement` value is I'm guessing that's a programming error somewhere where we aren't taking the absolute value of something. `-0.08` seems like much too large a value to occur "naturally" though accumulation of floating-point error. I took a quick look and it seems like `displacement` comes from the `segment_lengths`. There are both `linear` and `angular` segment lengths. The calculations for `linear` seem fine (impossible to be negative). However for angular in `firmware/app/control/trajectory_planner_impl.c:53` I suspect this is where the negative values are coming from. I'd bet adding `fabs` on this line would fix the issue. If you imagine a regular "1D" polynomial, I think this error occurs at the "top" of the curve where the first derivative flips from positive to negative (eg. when `x=0` for `y=-(x^2) + 3`). @garethellis0 please sanity check me here.

garethellis0, 2021-05-17:
I'll double check @MathewMacDougall's comment later, but just to confirm, you're not going to enable floating point exceptions for the actual binaries right? I'd recommend against it because it's unlikely that we'd catch all issues in simulated tests and segfaults during an actual game should be avoided at all costs.

garethellis0, 2021-05-17:
Another idea here is that you could make a 'tbots_gtest_main' target that enables the exceptions for you and then just have tests depend on that instead of the gtest_main target.

jonathanlew, 2021-05-17:
> I'll double check @MathewMacDougall's comment later, but just to confirm, you're not going to enable floating point exceptions for the actual binaries right? I'd recommend against it because it's [...]

Yes, that's right, we'd only enable the exceptions in tests

PR #2092 (2021-05-24): Fix Floating Point Error

## Please fill out the following before requesting review on this PR

### Description
We regularly have floating point errors in our codebase and that's not good for reasons discussed [here](https://github.com/UBC-Thunderbots/Software/issues/2088). 
* Added floating point error signals to all tests in the codebase
* Fixed all resulting errors
* converted simulated_test_main to tbots_gtest_main

### Testing Done
added floating point error detection to the common test main

### Resolved Issues
resolves #2088 

### Length Justification

### Review Checklist
closed 2021-05-24 by jonathanlew