UBC-Thunderbots/Software #2128: TrajectoryPlanner should calculate num_elements based on path length

MathewMacDougall, 2021-06-09:
### Description of the task

There is an implicit requirement that our robot controller/firmware runs **_at least_** as fast as the shortest `time_segment` in a `TrajectoryPlanner` planned trajectory.

[Currently for some given path (eg. the `MovePrimitive` moving to a destination), we create `PathParameters` with `num_segments=10`](https://github.com/UBC-Thunderbots/Software/blob/c71c01171ce697d077a3a0f17399eeb3cb195d85/src/firmware/app/primitives/move_primitive.c#L105). This will divide the path into 10 segments. Notably for control purposes, each segment corresponds to a single "control command", eg. accelerate, coast, decelerate. Therefore, shorter paths will have shorter segments, and therefore there will be less time between each segment.

If the firmware/control is running slower than the `TrajectoryPlanner` needs it to, [it will "skip ahead" segments until the current segment matches the current time](https://github.com/UBC-Thunderbots/Software/blob/c71c01171ce697d077a3a0f17399eeb3cb195d85/src/firmware/app/primitives/move_primitive.c#L134). This is so that we don't execute "old" control commands if we are already ahead. However skipping control commands can result in the robots moving very slowly, especially if we skip some "acceleration" or "coast" commands and decelerate instead.

This is what we observed currently in simulation with our custom simulator (on the latest `master`, commit `c71c01171ce697d077a3a0f17399eeb3cb195d85` at the time of writing).

We simulate our firmware at 200Hz. However for certain short motions, especially pivots, the difference in time between segments is sometimes `< 1/200 seconds`. This causes the robots to sometimes pivot really slowly.

To fix this, we need a "stronger" guarantee that we won't end up with time_segments smaller than the max speed we can run control. The ideal fix for this is likely to calculate the number of segments a path/trajectory should have, so that shorter paths get divided into fewer segments, avoiding tiny segments and therefore tiny unachievable timesteps.

### Acceptance criteria

- [ ] Calculate the num_segments for a trajectory planner path based on the path length

### Blocked By

MathewMacDougall, 2021-06-09:
This is what can happen currently. The robot pivots much slower than expected.
https://user-images.githubusercontent.com/21165407/121282873-13067180-c88f-11eb-861a-f01c6c9300ea.mp4

This is what should happen. Robots pivot much faster.
https://user-images.githubusercontent.com/21165407/121282869-0f72ea80-c88f-11eb-9364-9e7bf21c7c6f.mp4

MathewMacDougall, 2021-06-09:
https://user-images.githubusercontent.com/21165407/121282971-3a5d3e80-c88f-11eb-8dca-50e23a43b65a.mp4

jared-paul, 2021-06-09:
I think this might also be blocking #2108 a little bit? I noticed the pivoting was happening in spurts rather than being continuous. Not too sure if this was the cause though.

and like... do you already have a fix if you have the 2nd video?

MathewMacDougall, 2021-06-09:
Just put up https://github.com/UBC-Thunderbots/Software/pull/2129 with a "temporary" fix. The fix may be able to stay if it doesn't impact performance for too many people, but it should probably be fixed properly as well since who knows how fast real robot firmware will run.

MathewMacDougall, 2021-06-09:
Also to answer your original question, it _might_ affect #2108, but it depends exactly what the motion is.

closed 2021-06-17 by jonathanlew