UBC-Thunderbots/Software #2132: uint DynamicParams seem to fail compilation

MathewMacDougall, 2021-06-11:
### Description of the task

I tried changing the type of the `channel` param from `int` to `uint`, but then get the following error

```
bazel-out/k8-fastbuild/bin/shared/parameter/cpp_dynamic_parameters.h:3618:9: error: no match for 'operator=' (operand types are 'MutableParameterList {aka std::vector<std::variant<std::shared_ptr<Parameter<bool> >, std::shared_ptr<Parameter<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::shared_ptr<NumericParameter<int> >, std::shared_ptr<NumericParameter<double> >, std::shared_ptr<EnumeratedParameter<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::shared_ptr<Config> > >}' and '<brace-enclosed initializer list>')
         };
         ^
```

This occurs even when just building the `cpp_config_test` so isn't related to the types of the variables using the param in the actual code.

We should investigate this issue and then change the `channel` params in `network_config.yaml` and `full_system_main_command_line_args_config.yaml` to be `uint`

### Acceptance criteria

- [ ] Fix the compilation issue when parameters are of type `uint`
- [ ] Change both `channel` params to `uint`

### Blocked By

jonathanlew, 2021-06-11:
Please include the yaml snippet

MathewMacDougall, 2021-06-11:
```
- uint:
      name: channel
      min: 0
      max: 15
      value: 0
      description: >-
          The multicast channel to join to communicate with the robots

```

In `src/shared/parameter/config_definitions/full_system_main_command_line_args.yaml`

closed 2021-06-17 by jonathanlew